ThingPulse/esp8266-weather-station-color #1: Private Settings

anajantech, 2016-11-22:
I read your description of the Continuous Delivery Pipeline, and noticed how WiFi Manager allowed the router password to be removed from code.  Moreover, it looks like the router information is stored in flash, so it can be reused.

Have you thought about acquiring Weather Underground, ThingSpeak, and DST settings via browser connection (once WiFi is configured) and storing that information in flash too?  This would make the code completely portable.   I'm going to look into it, but it may take awhile with my coding skills (new).

jcruzp, 2016-12-22:
Dear friend there are an issue in file ArialRoundedMtBold_14.h change Mt by MT (ArialRoundedMTBold_14.h)

marcelstoer, 2018-04-14:
No activity in a long time -> closing

closed 2018-04-14 by marcelstoer