ThingPulse/esp8266-weather-station-color #5: U/L-case font include file problem

PuceBaboon, 2017-01-03:
Daniel,

   Thanks for publishing this neat project.  I used it as a quick test for a 320x240 ILI9341 connected to an ESP12 and was very impressed. 
   Like you, I'm now also a great fan of PlatformIO and use it for all my (mostly unfinished) projects.  I'm running under Linux, so there was/is one small problem with the build.  The Arial font library files are named inconsistently.  This probably doesn't affect Windows, but will cause builds under Unix or on Macs to fail.  The problem is with the small (14-point) filename, where the filename is written "Mt", but referred to in the #includes as "MT".
   It's easy enough for users to change the filename, but I just wanted to flag it, in case it trips up some novice maker somewhere.

   Many thanks from Japan and my best wishes for a happy and prosperous 2017,

                             -John-

squix78, 2017-07-15:
Thank you so much for reporting this. Can you confirm that it has been fixed and close this issue if so?

PuceBaboon, 2017-07-16:
Daniel,

   Sorry, I don't have much time to play right now, but my first quick
compile of the latest version fails with:-

gaijin@het:~/PlatformIO/Squix-Weather-Station-Colour$ pio run
[Sun Jul 16 09:56:15 2017] Processing esp12e (platform: espressif8266;
board: esp12e; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Converting esp8266-weather-station-color.ino
Collected 29 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <SPI> v1.0
|-- <ESP8266WiFi> v1.0
|-- <simpleDSTadjust> v1.2.0
|-- <WeatherStation> v1.3.1
|   |-- <JsonStreamingParser> v1.0.4
|   |-- <ESP8266WiFi> v1.0
|-- <Mini Grafx> v0.0.12
|   |-- <SPI> v1.0
|-- <JsonStreamingParser> v1.0.4
Compiling .pioenvs/esp12e/src/TouchController.o
Compiling .pioenvs/esp12e/src/esp8266-weather-station-color.ino.o
In file included from src/TouchController.cpp:1:0:
src/TouchController.h:2:33: fatal error: XPT2046_Touchscreen.h: No such
file or directory
#include <XPT2046_Touchscreen.h>
^
compilation terminated.
*** [.pioenvs/esp12e/src/TouchController.o] Error 1
/home/gaijin/PlatformIO/Squix-Weather-Station-Colour/src/esp8266-weather-station-color.ino:24:33:
fatal error: XPT2046_Touchscreen.h: No such file or directory
#include <XPT2046_Touchscreen.h>
^
compilation terminated.
*** [.pioenvs/esp12e/src/esp8266-weather-station-color.ino.o] Error 1
================================================================== [ERROR]
Took 1.08 seconds
==================================================================

   As you can see, your recommended libraries are installed.

                              -John-

PuceBaboon, 2017-07-18:
Daniel,

   Okay, a little more time to breathe today.  :-)

   Loaded library #542 (Paul Stoffregen's version of
the XPT2046_Touchscreen lib) and the compile succeeded.

   Looks like the font issue has been fixed with your integration of the
fonts into the one file, so looks okay on Unix-like systems now.

   Thanks!

                            -John-

closed 2017-07-22 by squix78