ThingPulse/esp8266-weather-station-color #6: Please combine WU API calls

taogeek, 2017-01-07:


G6EJD, 2017-01-07:
Yes but this increases the processing complexity and separating out the three responses, does it really make it any more efficient?

taogeek, 2017-01-07:
I just did it as a test, and yes, I think it's worthwhile in this case.  

There is no change to processing the results.  All code stays the same.  But instead of 3 calls for data, it's 1.  And it's two lines in the WU code -- easy enough that I could do it!

squix78, 2017-07-15:
Thank you very much for this suggestion. I decided to go another way and actually split the different API endpoints in separate classes due to memory problems and for cleaning the code.  Hence I will close this issue as "Won't fix". I hope you agree

closed 2017-07-15 by squix78

pilnikov, 2018-05-16:
astronomy and forecast prognosys request can do once time per day. Only condition need update after 15-30 minutes. 
`updateData(bool conditions, bool astronomy, bool forecast);`
And a call with conditions

G6EJD, 2018-05-16:
the forecast data is updated at regular intervals throughout the day, but updating astronomy only once/day is a good idea.