ThingPulse/esp8266-weather-station-color #10: doesn't display any icons any more

daduke, 2017-03-28:
Hi,

thanks for this great project! I built a weather station a couple of months ago and it's been working great until about 4 days ago. Now it doesn't display any icons. The text is fine, but no bmps. On the serial terminal I see that they get downloaded and also load from the file system, but they don't get displayed. Any hints?

thanks,
-Christian

daduke, 2017-03-28:
ah just saw the PR now, will try that. Strange that it's been working until recently..

G6EJD, 2017-03-29:
I've noticed that when Adafruit recently updated their GFX library my projects can no-longer display BMP Images, but the problem is solved by going back to GFX v1.02, I have not yet inspected the code differences between 1.02 and 1.03 (latest) to see why.

dronecz, 2017-04-09:
Hi all, 

I searched Adafruit forum and found this topic: [https://forums.adafruit.com/viewtopic.php?f=53&t=114248&p=570812&hilit=weather+station#p570812](url) and you have to update GfxUi.cpp in your sketch. I did test on 2 boards and if problem is not solved for the first time, you have to format SPIFFS (uncomment SPIFFS.format(); in main file of the sketch) and download all BMP files again.  

You can download file from here ;-), just rename it fromGfxUi.cpp.txt to GfxUi.cpp and you good to go. 
[GfxUi.cpp.txt](https://github.com/squix78/esp8266-weather-station-color/files/908600/GfxUi.cpp.txt)

Good luck ;-)

johnym30, 2017-04-15:
I still get no icons with the new cpp file and after spiffs format.Is there something else i could try?

dronecz, 2017-04-15:
Hi, you can try to force download all icons again. It is somewhere in main
sketch, near SPIFFS format I think.

G6EJD, 2017-04-15:
Have you tried downgrading <Adafruit_ILI9341.h> to 1.0.2 my copy of v1.0.6 functions correctly now it's been corrected.  I'm currently using:
#include <Adafruit_GFX.h>      // v1.1.8 
#include <Adafruit_ILI9341.h> // v1.0.6 

Also make sure the bmp draw function has 16-bit parameters ...String filename, uint16_t x, uint16_t y)

These are the areas I check for when BMP's stop drawing.

squix78, 2017-07-15:
Hi all. Thank you so much for reporting all this valuable information. Do you agree that we can close this issue since there is no more need for downloading icons and everything is based on the MiniGrafx library instead?

squix78, 2017-07-22:
Closing due to no response after one week

closed 2017-07-22 by squix78