ThingPulse/esp8266-weather-station-color #11: Display Problem....

njordan77, 2017-04-30:
Did setup everything today....seems to work as the BMPs are uploaded and I can see day+time....

But icons are not recognizeable at all....i can see the Temp correctly...I also can see SUN 5/15 MON 7/17 and TUE 7/17.....so very strange.....the dates look strange....but the TOP date with time is correct...

At the bottom i can see SUN and Rise/set time....also MOON and two values....but no icon for both of them....reset does not change anything.....

Any idea what is wrong....

G6EJD, 2017-04-30:
The adafruit libraries can be downgraded to fix this problem, or check your libraries are at the latest revision first there is a problem in the draw but map function

G6EJD, 2017-04-30:
I recall version 1.0.6 works (adafruit)

njordan77, 2017-04-30:
I have 1.06 installed, all libs are in latest version.
also strange that the weather preview has completely wrong dates....and complete weird pixels showing which do not give any sense.....only date time on TOP works fine.

njordan77, 2017-04-30:
now tried some previous version of Adafruit libs, but same results. weird graphics....how can I enforce loading of bmp's again....as after first load and reflash they are not loaded another time.

njordan77, 2017-04-30:
fixed in now, all libs in latest version. only took the GFX cpp from another issue threat.
Regards

squix78, 2017-07-22:
Closing since njordan77 confirms it is fixed

closed 2017-07-22 by squix78