ThingPulse/esp8266-weather-station-color #12: converting to non-pointer type 'int' from NULL in GfxUi.cpp

awumaker, 2017-05-02:
It seems, that SPIFFS.open(filename, "r") returns 0 and not NULL, if it can't find the file.

At least my compiler gives me a WARNING:

src\GfxUi.cpp: In member function 'void GfxUi::drawBmp(String, uint8_t, uint16_t)':
src\GfxUi.cpp:107:49: warning: converting to non-pointer type 'int' from NULL [-Wconversion-null]
if ((bmpFile = SPIFFS.open(filename, "r")) == NULL) {
^

and when I change it to:

  if ((bmpFile = SPIFFS.open(filename, "r")) == 0){
    Serial.print(F("File not found"));
    return;
  }

the compiler is happy and shows no WARNING.

Maybe it helps, to get the code with a warning less...

squix78, 2017-07-15:
This part of the code has been removed with the new version based on the MiniGrafx library. Do you agree that we can close this issue?

squix78, 2017-07-22:
Closing due to no response after one week

closed 2017-07-22 by squix78