ThingPulse/esp8266-weather-station-color #14: chancetstorms.bmp

Andrew252010, 2017-05-08:
http://www.squix.org/blog/wunderground/chancetstorms.bmp There is nothing at this address. Does this icon exist? Instead, it displays an empty space.

G6EJD, 2017-05-08:
It exists on the weather underground server so missing in Daniels folder

spiff72, 2017-05-27:
I just discovered this today, also.

I found that there is a file called "chancestorms.bmp".  I was able add that item to the list in the settings.h file, and it downloaded the file, but it won't work in the program because it is trying to use the name "chancetstorms" from wunderground.
`String wundergroundIcons [] = {"chanceflurries","chancerain","chancesleet","chancesnow","chancestorms","clear","cloudy","flurries","fog","hazy","mostlycloudy","mostlysunny","partlycloudy","partlysunny","rain","sleet","snow","sunny","tstorms","unknown"};
`

spiff72, 2017-05-28:
I think I fixed this by using the "chancestorms" text shown in the above line, and I corrected the line in the getMeteoconIcon section near the end of the sketch:

  if (iconText == "S") return "chancestorms";

G6EJD, 2017-05-28:
Except when there is a 'Chance of Thunder Storms' you'll get 'Chance of Storms' icon which would be incorrect. https://icons.wxug.com/i/c/a/chancetstorms.gif refers to the correct icon. It seems to be missing from your downloads.

G6EJD, 2017-05-28:
Depending on your chosen icon set ('a' from abcdefghijk') in my example, you could download it, then convert to BMP then upload to your SPIFFS, ideally Daniel would add it to his folder as it is missing.

spiff72, 2017-05-28:
I think this is just a typo in the code (where the "t" got dropped from the name).  From what I can see from looking at the various icon sets, there is no "chancestorms" icon in any of the sets.  There is only a "chancetstorms" icon.

So https://icons.wxug.com/i/c/i/chancetstorms.gif links to an icon, but https://icons.wxug.com/i/c/i/chancestorms.gif leads to a "not found".

So I think my fix described above is still valid.  It is just making an adjustment to point to an incorrectly named file on Daniel's server.

marcelstoer, 2018-04-14:
No longer relevant -> closing

closed 2018-04-14 by marcelstoer