ThingPulse/esp8266-weather-station-color #15: 'D1' was not declared in this scope

briankdlee, 2017-05-22:
I checked out latest version(#e71c540). 
And I got this error while building.
How can I fix this?

(I upgraded libraries on arduino library manager, and removed Adafruit ILI9341 library in arduino libraries folder)

`
In file included from D:\Projects\ESP8266\esp8266-weather-station-color\esp8266-weather-station-color.ino:46:0:

settings.h:26: error: 'D1' was not declared in this scope

 #define TFT_CS D1

                ^

D:\Projects\ESP8266\esp8266-weather-station-color\esp8266-weather-station-color.ino:58:41: note: in expansion of macro 'TFT_CS'

 Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC);

                                         ^

settings.h:25: error: 'D2' was not declared in this scope

 #define TFT_DC D2

                ^

D:\Projects\ESP8266\esp8266-weather-station-color\esp8266-weather-station-color.ino:58:49: note: in expansion of macro 'TFT_DC'

 Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC);

                                                 ^

D:\Projects\ESP8266\esp8266-weather-station-color\esp8266-weather-station-color.ino: In function 'void setup()':

settings.h:27: error: 'D8' was not declared in this scope

 #define LED_PIN D8

                 ^

D:\Projects\ESP8266\esp8266-weather-station-color\esp8266-weather-station-color.ino:90:11: note: in expansion of macro 'LED_PIN'

   pinMode(LED_PIN, D8);

           ^

exit status 1
'D1' was not declared in this scope`

G6EJD, 2017-05-23:
What board (ESP) have you selected in the IDE, you are getting these errors because pins D1 etc have not been defined, so define D1 and any others like D8 with your chosen device data pins.

briankdlee, 2017-05-23:
I attempt to download this firmware into NodeMCU 1.0.
I changed the board from 'Generic ESP8266 Module' to NodeMCU 1.0(ESP-12E Module)' in Arduino IDE.
And the above error does not happened.

Thanks a lot.

closed 2017-05-23 by briankdlee

jbishop129, 2018-09-19:
Resolved. As usual, the sure-fire way to fix a bug that eludes you: Spend hours trying to figure it out, then ultimately post an Issue; shortly thereafter, you'll find the solution :)

To that end, the problem was I had the wrong board selected in the IDE. I changed to the correct board, "NodeMCU 1.0 (ESP-12E Module)", and this resolved my issue. Code is working, stepper motors are running! Can close this issue report.

marcelstoer, 2018-09-19:
Guess you meant to post this here https://github.com/nidayand/motor-on-roller-blind-ws/issues/18 instead