ThingPulse/esp8266-weather-station-color #16: Save images locally

AFUDIrk, 2017-05-26:
Hello, the display ILI9341 has a slot for an SD card. If you can not use this option,
To put the images for the weather and the moon phases there?
Then they do not need to be loaded from the Internet every time.

greeting
Dirk

KarlNorway, 2017-05-26:
They are only downloaded if the are not present in SPIFF

G6EJD, 2017-05-26:
You could use the isplays SD card, but the additional wiring and need for another data pin for SD card chip-select makes using the on-board storage an elegant solution especially as the chip's flash storage is ample and generally unused.

AFUDIrk, 2017-05-28:
True, in the new version, the images are stored in the SPIFF and it works. It has only a small error, at chancetstorms.bmp not chancestorms and the folder under recourses must be mini and not small.

closed 2017-07-10 by AFUDIrk