ThingPulse/esp8266-weather-station-color #19: Displaying of actual indoor data from DHT22

vadpol, 2017-07-03:
Just thought that it might be useful to attach a DHT22 sensor to still free D0 (GPIO16) port in order to measure an actual room temperature and humidity and to display it at 3rd carousel frame for example. Or to add a still reserved touch screen functionalty (XPT) in order to trigger in between carousel forecast frames and fixed indoor conditions frame. It might be also good to indicate a city name somewhere on the screen in order to know which place weather it is. Also miss bmp moonphases in the new release, The present graphic ones look quite simple....  I think that all of above will increase functionality and  look and feel. Is it possible to update the code this way? Many thanks!

squix78, 2017-07-03:
The DHT22 should be easy to attach. The touch screen is also on my backlog and the touch demo in the mini grafx library already shows how it can be used: https://github.com/squix78/minigrafx/tree/master/examples/ILI9341/TouchPaint

City name is also a very good idea! The moon phase bmp is actually a technical issue I could not overcome yet. Problem there is that the weather station now is flicker free but has to live with only 4 colors. So bmps with more colors are problematic...

vadpol, 2017-07-04:
Thanks for your fast reaction! Actually my problem at the moment is to display the indoor climate data in a good way. A DHT22 sensor itself and its connection is not a problem indeed, I am looking for its data displaying solution on a given color display as a part of the present code as per above.  It would be of a great help if you could make it as you are familiar well with this code and it will take time for me to dig in.  Thank you very much!

smithbrandt, 2017-07-07:
Adding city name was pretty easy.

Add the following to the declarations of the ino file:

String myLocation = WUNDERGROUND_CITY;

I modified the drawCurrentWeather function to move the current conditions text under the current temperature on the display. Then, I put the city where the current conditions text was.

Add the following to the drawCurrentWeather function:

`// draws current weather information
void drawCurrentWeather() {
  gfx.setTransparentColor(MINI_BLACK);
  gfx.drawPalettedBitmapFromPgm(0, 55, getMeteoconIconFromProgmem(conditions.weatherIcon));

  // Weather Location
  myLoc.replace("_"," ");
  gfx.setFont(ArialRoundedMTBold_14);
  gfx.setColor(MINI_BLUE);
  gfx.setTextAlignment(TEXT_ALIGN_RIGHT);
  gfx.drawString(220, 76, myLoc);

  gfx.setFont(ArialRoundedMTBold_36);
  gfx.setColor(MINI_WHITE);
  gfx.setTextAlignment(TEXT_ALIGN_RIGHT);
  String degreeSign = "°F";
  if (IS_METRIC) {
    degreeSign = "°C";
  }
  String temp = conditions.currentTemp + degreeSign;
  gfx.drawString(220, 89, temp);

  // Weather Text
  gfx.setFont(ArialRoundedMTBold_14);
  gfx.setColor(MINI_YELLOW);
  gfx.setTextAlignment(TEXT_ALIGN_RIGHT);
  gfx.drawString(220, 125, conditions.weatherText);

}`

vadpol, 2017-07-09:
Yes, this works pretty nice, just small remark: the declaration should be

String myLoc = WUNDERGROUND_CITY;

otherwise the given in above is not recognized.

I am still struggling with touch screen functionality as the above demo link doesn't work for me. When I upload the Touch demo provided it shows immediately the color pallet at the top of the screen and a small white triangle dot in the left bottom corner of the screen, no ask to touch the screen is displayed. The screen doesn't react on any touch however the blue LED on ESP module goes bright when screen is touched. All other demos like Rotating Cube and Weather station color itself work just fine. Perhaps the problem is with XPT2046_Touchscreen.h library. I used this one 

https://github.com/PaulStoffregen/XPT2046_Touchscreen

Any idea on what is wrong? Many thanks!

smithbrandt, 2017-07-09:
You're right! I typed the first part and changed myLoc to myLocation, and then copied and pasted my code but didn't change the variable name.

squix78, 2017-07-15:
I added both city name and touch screen functionality, although some people have problems with it. Can this issue be closed?

vadpol, 2017-07-16:
Hi Daniel,

I think so, thanks for your effort! 

I just experimented with BMP280 SPI sensor to measure the indoor climate and it works perfectly well in combination with DHT22. I've added the 6th info screen which shows now:

1.  Indoor temperature in Celsius - measured by DHT22 sensor
2.  Relative humidity in the room in % - measured by DHT22 sensor
3.  Air pressure in mm Hg - measured by BMP180 sensor
4.  Approximate altitude above the sea level in m - measured by BMP180 sensor

This is done at the cost of connecting of LED pin to +3,3V power source via 100 Ohm resistor. This allowed me to have 2 free GPIO's needed for CS signal of the both sensors (DHT22 and BMP280). The rest signal pins of them are connected to common hardware SPI bus. 

For the next improvement I ordered a BME280 SPI sensor which could measure all the above parameters alone, therefore it will be no need in DHT22 sensor anymore. This way it will be possible again to regulate the screen brightness via GPIO15 (D8) pin.

Unfortunately the touch screen functionality doesn't work for me well - all info screens are scrolling very fast without any touching, also it could be a display problem (XPT chip failure for example) or so. 

Nevertheless I close the issue and will try to fight with this problem on my own, I see some ways how to do it, some experiments are needed though.

As soon as I manage to make a final design I will post it.

Best regards,
Vadim

closed 2017-07-16 by vadpol

anthonyjclarke, 2017-07-16:
@vadpol would love to borrow your code.. been meaning to do this for a while... :)

AFUDIrk, 2017-07-17:
Hello vadpol, it would be nice if you could publish the code, am very interested in it.
I've already times thought about the current inside and outside temperature on the weather station display, but unfortunately I know not yet so well from me the programming.

vadpol, 2017-07-18:
It seems that I've found the possible reason of screen scrolling with XPT2046. As to me it happens just because the onboard +3,3v regualator on D1 Mini ESP module is not enough to power everything in this setup especially LED backlight of the display. It works at the edge of its capacity I believe and some of the XPT chips might be very sensitive to voltage fluctuations. I disconnected an LED pin at the display from GPIO and connected to +3,3v pin at D1 Mini module via 100 Ohm resistor first. It doesn't help a lot as the regulator was still loaded. Then I connected LED pin via the same resistor to +5V pin at D1, so directly to USB input power. And screen scrolling has stopped! Everything works now how it meant to work! If I touch the screen it goes to next one, if not -  it shows the current screen with refreshed from time to time info. Hope that this might be the reason, interesting to know from others whether it helps as well. 
PS I will publish my code update soon.

AFUDIrk, 2017-11-07:
Hallo vadpol, es ist schon eine Zeit her, hast Du den Code mittlerweile fertig, bin sehr daran interessiert.

Thermisto, 2017-11-07:
I have followed the instructions in this [thread](https://github.com/squix78/esp8266-weather-station-color/issues/24) and it works very well for me. No issues with touch.

mpo881, 2018-11-25:
@vadpol any update? Id love to see how you handle connecting either the dht-22 or bme280
Thanks!