ThingPulse/esp8266-weather-station-color #21: Can't compile

Trecoolius, 2017-07-10:
```
esp8266-weather-station-color:524: error: 'class MiniGrafx' has no member named 'drawStringMaxWidth'

   gfx.drawStringMaxWidth(15, 265, 240 - 2 * 15, ESP.getResetInfo());

       ^

exit status 1
'class MiniGrafx' has no member named 'drawStringMaxWidth'
```

squix78, 2017-07-10:
Please update the MiniGrafx library to version 0.0.11. It could be that the
Arduino IDE library manager does not offer this version yet from the drop
down. Then either wait and check back or download the library from the
github repository manually and replace the already installed version:
https://github.com/squix78/minigrafx

Please close this issue if you agree that the library update solves it

Thank you
Daniel

Trecoolius, 2017-07-10:
That did it, thanks.

closed 2017-07-10 by Trecoolius