ThingPulse/esp8266-weather-station-color #22: Info screens go too fast

Trecoolius, 2017-07-10:
Got it working, but it seems the script is not retrieving the info from wunderground, since the weather shows 0°C and the screens cycle super fast. API key was set and manual requests from my browser display all the info needed (eg http://api.wunderground.com/api/MY API KEY/conditions/q/USA/NEW_YORK.json).

squix78, 2017-07-10:
Please try to replace USA with NY:
http://api.wunderground.com/api/YOUR_API_KEY/conditions/q/NY/NEW_YORK.json
When I tried the call with USA then I got a list of possible candidate
locations...

G6EJD, 2017-07-10:
One day WU will publish details of their API's including allowed values, I asked them if they would do this, but they said the current supplied documentation is adequate. For example I discovered their systems sometimes report wind direction as WEST when other days it usually reports 'W' only, there's an example where API values like County and City values are needed.

Trecoolius, 2017-07-10:
Awesome, works now, but the screen go by too fast. Is there any setting I can change to make them go slower?

squix78, 2017-07-10:
Are they moving automatically? They actually should not move at all until
you press the touch screen, assuming that you have a display module with a
touch screen (e.g. from my ESP8266 Wifi Color Display Kit
https://blog.squix.org/product/esp8266-wifi-color-display-kit-2-4). If you
are talking about the forecast slider on the first screen you can control
this by calling carousel.setTimePerFrame(time)

Trecoolius, 2017-07-10:
It's not the starting slider, it's everything :(

squix78, 2017-07-10:
What display are you using? Does it have touch screen? If not I recommend
to comment the line "screen = (screen + 1) % screenCount;"
Then you can't switch from frame to frame but you can always add a button
or so

Trecoolius, 2017-07-10:
I'm using [this one](http://www.ebay.com/itm/2-8-TFT-LCD-Display-Touch-Panel-SPI-Serial-240-320-ILI9341-5V-3-3V-STM32/201950756171).

squix78, 2017-07-10:
I have one like that but have not tried it with the touch interface...

marcosvidalfacal, 2017-07-12:
Hi all

I'm also seeing the same issue on my display.
I have this display [https://learn.adafruit.com/wifi-weather-station-with-tft-display/overview].

Regards,
Marcos

HaxxHaris, 2017-07-12:
Hi, 

I got my screen today and I have same issue with my display 2.4" TFT LCD Display Touch Panel SPI Serial 240*320 ILI9341 

I have tried to disconnect pins for touchscreen but it did not help. The screen changes automatically. From page 1 to 5 in maybe 2-3 seconds

Unfortunately, I can not program so I do not know where to enter carousel.setTimePerFrame (time) exactly :/

vadpol, 2017-07-13:
I report the same issue, I have a 2,8" TFT LCD SPI 240*320 ILI9321 display with XPT2046 Touch Screen chip. After flashing of the code and connection to WiFi all 5 screens start automatic rapid scrolling one after other in majority cases also after reset button is pressed.  However in a very seldom case one of the random screens stops, but automatic scrolling starts again as soon as screen is touched. I found that when scrolling happens the blue LED on the module is blinking very fast, when screen is stable the LED is off and it goes bright while screen is touched, doesn't matter stable or scrolling screens are present at the moment.

squix78, 2017-07-14:
Thank you all for your feedback. Are you sure that the 2.8" display uses a XPT2046 touch screen chip? It sounds like that the described behavior occurs, when the touch driver has not been detected properly and they isTouched method always returns true. This should be certainly fixed on the software side. To actually solve your problem I would like to hear from you if it might be possible that the 2.8" display uses a different driver chip

marcosvidalfacal, 2017-07-14:
Hello

In my case you are right.
The controller of my touch unit is a STMPE610 have a look at page 14 in [this](https://cdn-learn.adafruit.com/downloads/pdf/adafruit-2-4-tft-touch-screen-featherwing.pdf) doc.

Thanks a lot for the hint.

Regards,
Marcos

vadpol, 2017-07-14:
In my case it is XPT2046
[image]

HaxxHaris, 2017-07-14:
Same here XPT2046 ours may be are fake IC :)
[image]

marcosvidalfacal, 2017-07-14:
Hello
I managed to get my display/touch starting working the expected way.
I downloaded and intslled the Adafruit [lib](https://github.com/adafruit/Adafruit_STMPE610).
I modified the code to disable the lib for the XPT2046 and replaced parts of the code to use the Adafruit lib.
Currently the pages swap only after a touch on the screen.
Switching from 24h to 12h display is not working as the code is not able yet to detect where the user touched the screen.
I will try to get this issue also solved.
I will also try to get the display switched off after a couple of seconds and switch it on at the first screen touch.
I still do not understand why the red led is blinking all the time, any hint on this?

Bye
Marcos

vadpol, 2017-07-18:
It seems that I've found the possible reason of screen scrolling with XPT2046. As to me it happens just because the onboard +3,3v regualator on D1 Mini ESP module is not enough to power everything in this setup especially LED backlight of the display. It works at the edge of its capacity I believe and some of the XPT chips might be very sensitive to voltage fluctuations. I disconnected an LED pin at the display from GPIO and connected to +3,3v pin at D1 Mini module via 100 Ohm resistor first. It doesn't help a lot as the regulator was still loaded. Then I connected LED pin via the same resistor to +5V pin at D1, so directly to USB input power. And screen scrolling has stopped! Everything works now how it meant to work! If I touch the screen it goes to next one, if not -  it shows the current screen with refreshed from time to time info. Hope that this might be the reason, interesting to know from others whether it helps as well.

scropion86, 2017-08-14:
@vadpol can you share the modified Lib which working for you , so we can test it .
i have identical LCD screen and same issue of touch problem.

vadpol, 2017-08-15:
Hi,

I didn't change any lib, the thing worked for me is just re-connection of the display's backlight LED pin to +5V power (USB) via 100 Ohm resistor. This allowed to take out a part of the load from 3,3V regulator at D1 Mini board and thus influented XPT somehow. I still have no idea why it is so, but it worked for me. I have made some improvements of this schematics later on, you may find my the latest developments under the Screen brightness issue up here.

scropion86, 2017-08-23:
the problem still not solved for me with the above solution that @vadpol purposed
i even try to provide a 3.3V from other source and share the ground to take out the entire load from the USB.
any other work around??

vadpol, 2017-08-28:
I also tried to put 1000,0 x 6,3V and 0,1uF capacitors at both power lines (+3,3V and +5V) thinking that it might be some LF and/or HF distortions, however no result as well. By the way I never saw a calibration screen appeared and prompt to switch in between 12H /24H and METRIC/IMPERIAL modes. Trying of the separate touch screen test examples also didn't work.  So my conclusion at the moment is that some batches of XPT2046 are not good enough or even fake indeed and this is a pure hardware issue. Different XPT2046 chips behave differently, there is no system in it...

scropion86, 2017-08-28:
@vadpol i so sad to say that i already found the problem , there was a small crack on the touch glass that's make problem.
i figured it out by trying a sketch for touch painting which automatically fill the screen line by line..
the crack was on the corner small unseen and later it spread diagonally enough see it

check you screen for crack 😢

vadpol, 2017-08-28:
Well, how did you fix it then? I think it is not an easy thing....

scropion86, 2017-08-28:
no way i just saved my time and ordered a new screen with less headache i ordered [nextion screen 2.8"](https://www.itead.cc/nextion-nx3224t028-1931.html) they have a good enough IDE to design the GUI
[here ](https://nextion.itead.cc/)More about

kiwipaulrob, 2017-10-22:
I was tearing my hair out trying to get my display to work when I found out that I had a display with no touch sensor. See photos. I ordered another that is identical except it has the resistive touch sensor in place. See the extra connector on the left.

https://photos.app.goo.gl/CHiLMIcjrS65UYXm1

njordan77, 2017-11-05:
Hi Daniel, was this issue discussed any further. I have several 2,4/2.,8/3,2 inch TFT touch displays with the named chip (fake?) but non of them works with touch.

In my case strange, when first starting i shortly could see a calibration screen which immediately started to switch like with others every 2-3 seconds....
After clearing the flash and reinstallation of the same code without any adaptation i now have the strange situation that no calibration request (even if it did not work) and also no rotation. Now the main screen is kept, only the 12/24 hour clock switches every 2-3 seconds. Touch of course does not react. Wiring is the same as your plane wiring scheme.
Any further digging into this issue? Any idea why a full flush of flash does not bring the same starting situation with calibration request? (if full flash is cleared).

+ what would be necessary to disable the TOUCH part if it does not work at all. So i could put a simple button to get alternative switching.

Many Thanks, Norbert

mcouture87, 2017-12-15:
I have the TFT FeatherWing from AdaFruit 2.4" touchscreen and I got the same issues.   First boot, got the calibration screen, then it started to flip through all the screens very quickly.   On all future boots, I never get the calibration screen anymore but all screens flip by very quickly.

rberkelm, 2018-01-20:
Has there been any progress on resolving this issue? I bought my new weather station kit from the Squix 
online shop just before Xmas, complete with 2.4" TFT screen. Worked beautifully for the first few days. Then, after a reboot, the screens keep flipping by very quickly. Touching the screen does not help. Reboots don't help. Would dearly like to find a solution.

lorol, 2018-01-20:
This mostly is caused by 2 things 
-  factory defective or later failed screen module (touching layer or circuit) 
-  Since LED back-light is powered directly from ESP processor pin,  the load is too high and 3V3 can be unstable for the requirements of screen module touch sensing circuit.

You can try another screen if you have, you can try cxandy's clone of the project - it has a button replacement of the touch, also you can see above comments for power cranking workarounds.

rberkelm, 2018-01-20:
Thanks lorol. Looks like a faulty screen for me as I'm powering from a 2A supply and have tried other power options. I've ordered another screen but might try the button option as well. I've tried slowing down the screens with delays and it works, but the clock updates don't happen as they should. Now I know what the "remainingTimeBudget" line and comment is about!

squix78, 2018-01-21:
Hi All. Sorry for being absent from this thread for such a long time. I'd love to find a solution to these problems but I don't have a setup to analyze these issues. If one of you could send me their screen I might be help. But without it it's just educated guessing

squix78, 2018-01-21:
Has anyone of you tried with a different library? These guy here says his white screen problem disappears when using the adafruit library: http://discuss.tindie.com/t/pinout-on-esp8266-display-kit/23871/3

rberkelm, 2018-01-21:
Hi Dani, I'll send you mine if you like. It came from your online store, so
you might be interested in getting this one sorted. Since you have done
such an amazing job on this project I'm happy to cover the freight. Can you
give me your postal details?
Ray

rberkelm, 2018-02-03:
I'm still not making progress with issue. I soldered a 1000uF capacitor onto the TFT VCC and GND pins thinking that perhaps the regulator on the Wemos board might be a bit borderline. No luck! 
Today I received a new 2.4" TFT screen and wired it up to my ESP dev board. It has the same issue with the fast-moving screen. It has a touch controller IC with HR2046 written on it. I've tried some Different libraries incl the Adafruit STMPE610, and Bodmer's TFT_Touch  and TFT_Touch_Shield_V2. None of these will compile for me.

UniquePete, 2018-04-15:
I am having similar problems with the touchscreen part of things—symptoms as previously described. My screen has an HR2406 controller, and the build quality is anything but outstanding, but I managed to get the display part functional by simply removing (commenting out) the 'if (touchController.isTouched(500))' conditional statement at around line 199. My unit also works a lot more reliably with an independent 5V power supply.

It would be nice to get to the bottom of the touchscreen problem though. It seems that whatever is going on, the MCU thinks that the screen is constantly being 'touched'. When running the TouchTest sketch, there is a constant stream of messages of to the Serial Monitor:

```
Pressure = 4095, x = 0, y = 1
Pressure = 4095, x = 0, y = 1
Pressure = 4095, x = 0, y = 1
Pressure = 4095, x = 0, y = 1
Pressure = 4095, x = 0, y = 1
Pressure = 4095, x = 0, y = 1
Pressure = 4095, x = 1, y = 1
Pressure = 4691, x = 611, y = 612
Pressure = 4655, x = 562, y = 576
Pressure = 4112, x = 211, y = 185
Pressure = 4098, x = 2, y = 1
Pressure = 4095, x = 0, y = 1
Pressure = 4095, x = 1, y = 1
Pressure = 4095, x = 0, y = 1
Pressure = 4095, x = 0, y = 1
Pressure = 4095, x = 0, y = 1
Pressure = 4095, x = 0, y = 1
```
Note that the screen is detecting my own tap above the 'background noise'—those are the entries with non-zero/one x & y values. Should we just be setting a sensitivity threshold somewhere?

UniquePete, 2018-04-15:
Well, I can get things partially working on my configuration by editing the XPT2046_Touchscreen.cpp file and setting the value of Z_THRESHOLD to something greater than the 'background' pressure measurement displayed when I ran the TouchTest sketch. I also set the Z_THRESHOLD_INT value correspondingly higher, but at this point I'm not sure what that is doing.
```
#define Z_THRESHOLD     4200
#define Z_THRESHOLD_INT	750
```
Now the esp8266-weather-station-color sketch will load and run, and I can change the time format by tapping on the screen. But that's all I can do. If I check the x & y values that are being returned, the y value is always 0, which explains why tapping anywhere on the screen only changes the time format. But the x value is sometimes negative, which doesn't make a lot of sense to me.

One other thing I noticed when running the TouchTest sketch was that, while the x & y values did vary as I tapped around the screen, I couldn't work out the logic of the x & y values being reported. In particular, the x & y values, apart from being affected in some way by the position being touched on the screen, were also very much affected by how hard I was pressing (i.e. if I just touched one point and pressed harder, the x & y values increased, significantly).

And why is the blue [NodeMCU] 'load LED' flashing constantly? (It's on hard as the TouchTest sketch is running too.)

Anyway, I'll fiddle some more. If anyone else works it out, please let us all know.

G6EJD, 2018-04-15:
The LED is flashing because on that board it is shared by one of the gpio ports in use you can determine which one by serial.println(BUILTIN_LED) if I remember the correct spelling. 

Have you tried running a touch test programme to eliminate any issues resulting from the hardware.

UniquePete, 2018-04-16:
Thanks for the tip. It wasn't that LED, but it was the same problem—the LED in question was linked to GPIO 2, which we were using for T_IRQ. Disconnect that lead and no more flashing LED. No more T_IRQ either, of course, but that didn't seem to make any difference...?!?

There aren't that many GPIO pins left to use though. Are we really using all those leads coming from the display? We only seem to be using four of them (CS, DC, T_CS and T_IRQ, notwithstanding my observation in relation to T_IRQ), plus the RST, LED and power leads I guess. Nonetheless, I tried GPIO9 (SD2), and that just resulted in a crash. GPIO10 (SD3) seems to be OK, so that's where T_IRQ is now, but given that completely disconnecting the T_IRQ line from the display seemed to have no impact on operations, I've no idea wether or not the current connection (to GPIO10) is doing anything.

And yes, the problem seems to exist when running the TouchTest sketch, so it's almost certainly hardware related. The question is really whether this is a simple configuration issue or a more intrinsic hardware problem.

The fact that the TouchTest output indicates that both x & y values are being sensed (even if it's not clear what they mean), while the weather station software constantly reports a y value of "0", suggests that there is still a software element to this issue.

UniquePete, 2018-04-16:
OK, so I appear to have fixed my remaining problem. Somewhere along the line, obviously when things weren't working properly, I had created a bad calibration file (valid file, but invalid data) that was being read every time I loaded the sketch. So I modified the sketch to set the result of the calibration check (at around line 170) to false and force a new calibration:
```
//  boolean isCalibrationAvailable = touchController.loadCalibration();
boolean isCalibrationAvailable = false;
```
After following the calibration instructions (creating the new calibration file), I changed the sketch code back to the original call to touchController.loadCalibration().

I still don't really understand the x & y values being returned, but now at least when I tap on the upper part of the display, the date format changes, and when I tap of the lower part of the display it cycles through several screens. I'm assuming that that's the correct behaviour...

mattie47, 2018-05-03:
Sadly also seeing the issue with all the screens switching through within a few seconds with this non-touch screen + D1 mini: https://www.aliexpress.com/item/Smart-Electronics-2-2-Inch-240-320-Dots-SPI-TFT-LCD-Serial-Port-Module-Display-ILI9341/32607741715.html

mattie47, 2018-08-02:
Anyone got an idea how to fix this?

As per https://github.com/ThingPulse/esp8266-weather-station-color/issues/52 it used to work prior to touch support. 

Thanks,

Matt

mattie47, 2018-08-13:
For anyone interested, you can use this with non-touch screens with the last commit before touch screen support was added. You don't get the fast scrolling issue.

For reference, compiling with this commit it works fine:

https://github.com/ThingPulse/esp8266-weather-station-color/tree/5ba7fa70b5b707a85b4a15fe4bdc9f9bc79b0d9c

👍

mattie47, 2018-08-17:
@commanzoom  Other than fixing the actual problem, you could always look at making the required changes for openweathermap at that git commit I referred to.

Openweathermap changes were done in https://github.com/ThingPulse/esp8266-weather-station-color/commit/5b6fef3da75738f050d128b6546ae56773dbd5bd

Sorry - I don't have time to give this a go, nor the C skills to try and fix/understand the problem with non-touch screens.

Thanks,

Matt

commanzoom, 2018-08-17:
Hi @mattie47 
Thanks for advice, but i was busy with this yesterday night and managed to fix this following advise from #84 - commented lines from 216 to 224 (master)

mattie47, 2018-08-19:
Thanks for that @commanzoom. I hadn't seen that post. Got mine working too. :-)

closed 2019-06-08 by stale[bot]

wapiti59, 2020-05-03:
Hi.  I'm sorry if this has been addressed.  I saw a reference to it up above but when I clicked on the link it would not let med view it.  I received my kit(s) directly from Thingpulse. got them all assembled and programmed no problem, and they boot up and run great - for about a minute, and then the screen goes completely white.  If I power down and back up again, works great, again for a minute and then white screen.  Help???