ThingPulse/esp8266-weather-station-color #23: Alerts Causing Crash

smithbrandt, 2017-07-13:
There have been sever weather alerts in my area the last two days and my weather station color will not boot. It crashes and reboots when it gets to the updating alerts section of the updates.

I commented out that portion of the code, and it boots fine. 

`/*
  drawProgress(90, "Updating alerts...");
  WundergroundAlerts *alertClient = new WundergroundAlerts();
  alertClient->updateAlerts(alerts, MAX_ALERTS, WUNDERGRROUND_API_KEY, WUNDERGRROUND_LANGUAGE, WUNDERGROUND_COUNTRY, WUNDERGROUND_CITY);
  delete alertClient;
  alertClient = nullptr;
*/`

I downloaded the json file from wunderground and it is 24KB in size. I have MAX_ALERTS set to 1.

Let me know if any additional info will help.

squix78, 2017-07-14:
Could you please post your JSON response from the alerts service here? I
couldn't find any city to test the alerts service

smithbrandt, 2017-07-14:
Here's the JSON file...

[Old_Bridge.zip](https://github.com/squix78/esp8266-weather-station-color/files/1148503/Old_Bridge.zip)

closed 2019-06-08 by stale[bot]