ThingPulse/esp8266-weather-station-color #36: Change language to German

djuscha, 2017-08-13:
Hab almost everything in German converted, only Forecast window shows me weather still in English .We are days in German. And in the main window above the date is in English how can I change that.
http://api.wunderground.com/api/apikey/forecast10day/lang:DL/q/Baunatal.jsonShows me the weather in German.
What am I doing wrong?
Everything else is in German and the comments I have also translated

Luc3as, 2017-09-19:
Hello, 
there is possibility now to translate the ui to other languages, see 
https://github.com/squix78/esp8266-weather-station-color/pull/43

djuscha, 2017-09-19:
Have tried still language_DE.h and language_RU.h to reiposten does not know whether you can embed that.

Luc3as, 2017-09-19:
If you do have translated the original file and saved it as language_de.h
the include definition is is main code file. Src/weather station color. Ino

djuscha, 2017-09-19:
eather-station-color.ino:55:25: fatal error: language_EN.h: No such file or directory

 #include <language_EN.h>

                         ^

compilation terminated.

djuscha, 2017-09-19:
have changed and now works switch between windows no more. even if I leave in English there is only one window.

Luc3as, 2017-09-20:
oh, I see, please uncomment line 222 //screen = (screen + 1) % screenCount;  in esp8266-weather-station-color.ino   I had problems with touchscreeen so I disabled rotating by touch and forget to put it back.

djuscha, 2017-09-20:
yes now But it is only komentare translated that I had already sketch translated directly. Im sketch also const string WUNDERGRROUND_LANGUAGE = "DL";
const String WUNDERGROUND_COUNTRY = "DE"; and still as I have written stands up in the first window day in English and weather display in the forestcast is still in English.also as before :(

Luc3as, 2017-09-20:
hmmm I think there is some strings hard coded in weather station library
which is used to download forecasts

djuscha, 2017-09-20:
Russian is not at all :(

Luc3as, 2017-09-20:
can you post photo of what is displaying wrong on display ?

djuscha, 2017-09-20:
https://github.com/djuscha/esp8266-weather-station-color
here are the pictures. De for German and RU for Cyril

Luc3as, 2017-09-20:
Okay, so the main problem here is the builtin font cannot show russian
characters at all, because there is just basic set of characters. solution
to this would be adding your russian charracters to the translit function,
which would change those charracters for latin characters. you can find
inspiration for characters to put to array here
https://stackoverflow.com/questions/7461406/cyrillic-transliteration-in-php

just keep in mind, after adding characters to the array, String lat_up[]
and so on, count elements in array and put that number to  for (int i = 0;
i < 11 ; i++) {

djuscha, 2017-09-20:
this is too difficult for me with russian :( .. the good I am not in the program
and what is with German? there is not translated everything.

Luc3as, 2017-09-20:
I updated the code for you, please try if it translating russian
charracters now, with german I think I cannot change it because it is hard
coded to library which we are using

djuscha, 2017-09-20:
have just tried with russian ... it is just like before, no change to see :(

Luc3as, 2017-09-20:
Sorry my fault. I forgot to update number in for loop statement

herrold, 2017-09-20:
@Luc3as said:
--- Just keep in mind, after adding characters to the array, String lat_up[] and so on, count elements in array and put that number to for (int i = 0; i < 11 ; i++)

computers are much more accurate than humans in counting -- is there not a 'length of' macro which can do this automatically at compile time?

herrold, 2017-09-20:
... sorry about all the markup changes -- I normally use another variant

Luc3as, 2017-09-20:
Yup I'm sure it is. Honestly I tried to get length somehow and it was not
working, more specifically it was crashing esp, so I put the number of
elements hard here. But it was just quick fix and I kept it this way. I
will look at it and do it as you suggest.

djuscha, 2017-09-20:
update number in for loop Statement  still not changed?

Luc3as, 2017-09-21:
Hello, I updated code to automatically count number of ellements in array, you can try

djuscha, 2017-09-21:
now what is displayed, but are russian words in translit, russian words with latin letters :(

Luc3as, 2017-09-21:
Correct, this is max I can do without knowing how to generate font files to
progmem.

djuscha, 2017-09-21:
oh so all clear, thanks for your work !!

closed 2018-04-15 by marcelstoer