ThingPulse/esp8266-weather-station-color #37: Sketch does not run

ericklil, 2017-08-21:
Hello, after installing on your WiFi Color Display `Kit` the sketch hangs her:

```
[HTTP] GET... code: 200
start document
[HTTP] GET...
[HTTP] GET... code: 200
start document

DST Rules Updated:
DST Start: Sun Mar 12 02:00:00 2017
DST End:   Sun Nov 05 01:00:00 2017
```

Screen displays nothing has a slight flicker ( also when plugged in to adequate power supply)
Any ideas would be welcomed..

Great project though!!

squix78, 2017-08-21:
Did you already have a look at questions and solutions here:
https://support.squix.org

ericklil, 2017-08-21:
Thanks Daniel,

i did but did not see a similar issue. i have to admit however that I did not read all posts and answers…
any input would be greatly appreciated ( all  samples run fine)

  Erick Oeseburg
       Encinitas
       92024 CA
  +1 (619) 988-1543
poolworks@mac.com

ericklil, 2017-08-21:
Sorry my bad some connections were not soldered properly....Thanks for your help!

closed 2017-08-21 by ericklil