ThingPulse/esp8266-weather-station-color #38: Additional Unicode Blocks in Oleddisplay tool

LaurisGale, 2017-08-29:
Would it be possible to have additional Unicode Blocks (e.g. Latin Extended-A/B, Arabic, Cyrillic, etc. subsets) in your oleddisplay.squix.ch tool? I understand that not all fonts have all subsets, but at least the ones that have (Roboto, opensans, Noto, etc have full subsets) would help to localize the weather tool. Thanks!

Luc3as, 2017-09-19:
Hello, 
I just pushed updated code to "fool" international characters, 
https://github.com/squix78/esp8266-weather-station-color/pull/43

LaurisGale, 2017-09-21:
That is great! I reused your translit code - works like a charm, but real characters would be even better :)

Luc3as, 2017-09-21:
I definitelly agree, but making extended font is something way beyond my skill now and as the author is currently not having time I guess the only way was like that. By the way If you have some additional translation files with strings translated you can post it somewhere and I can add it to the pull request.

LaurisGale, 2017-09-21:
Attached is the language_LT.h (Lithuanian translation). 

Letter replacement string: 

`
String Translit(String str) {
    String lat_up[] = {"Ą" , "Č" , "Ę" , "Ė" , "Į" , "Š" , "Ų" , "Ū" , "Ž"};
    String lat_low[] = {"ą" , "č" , "ę" , "ė" , "į" , "š" , "ų" , "ū" , "ž"};
    String trans_up[] = {"A" , "C" , "E" , "E" , "I" , "S" , "U" , "U" , "Z"};
    String trans_low[] = {"a" , "c" , "e" , "e" , "i" , "s" , "u" , "u" , "z"};
    for (int i = 0; i < 9 ; i++) {
      str.replace(lat_up[i],trans_up[i]);
      str.replace(lat_low[i],trans_low[i]);
    }
    return str;
}
`

[language_LT.h.txt](https://github.com/squix78/esp8266-weather-station-color/files/1321950/language_LT.h.txt)

LaurisGale, 2017-09-21:
There is also one thing that does not want to translit. The helper for forecast columns where weekdays are truncated to 3 chars and I cannot correctly transliterate it. Tried differently but here's the latest code:

// helper for the forecast columns
void drawForecastDetail(uint16_t x, uint16_t y, uint8_t dayIndex) {
  gfx.setColor(MINI_YELLOW);
  gfx.setFont(Roboto_Medium_14);
  gfx.setTextAlignment(TEXT_ALIGN_CENTER);
  String day = **Translit**(forecasts[dayIndex].forecastTitle.substring(0, 3));
  day.toUpperCase();
  gfx.drawString(x + 25, y - 15, **Translit**(day));

Any thoughts?

Luc3as, 2017-09-21:
I have similar problem with Thursday, I think it is because In special
characters there are more bits used to store the letter and If it clip it
at 3 first letters (or bits or something) it is moved beyond.
Maybe using transliteration first and then clipping to 3 letters would
help.
Have to try

closed 2019-06-08 by stale[bot]