ThingPulse/esp8266-weather-station-color #39: Forecast last entry has wrong icon

ccpetersen, 2017-09-05:
On the last page of the Forecast display the last entry has an icon, cloud with ? mark which is not correct.

G6EJD, 2017-09-05:
I thought the Weather Underground icons were used, in which case they do have clouds with a question mark to denote unknown weather, see here: https://www.wunderground.com/weather/api/d/docs?d=resources/icon-sets

ccpetersen, 2017-09-05:
It's always the last entry on the second "Forecast" page and it never varies.  The text description for that entry is also null. I have another project using the json data for this local and its "Forecast" is correct so I suspect it's not being parsed correctly here.  I'll have a look at the code later today.

BTW the weather icons used are in weathericons.h, not the WU icons.

ccpetersen, 2017-09-10:
So far I've found that forecasts[i].forecastIcon returns "nt_clear" for the last entry.  getMiniMeteoconIconFromProgmem() and GetShortText() don't know how to handle that entry.  If you add "nt_clear" to these functions and return "clear" and "Clear" respectively it works.  

Have to find out why "nt_clear" is returned.

lorol, 2017-09-16:
ccpetersen,  thanks for the analyses. nt_xxxxxx  seem to be for "night time". You can either add nt_ support resources or cut the nt_

ccpetersen, 2017-09-18:
All entires on the forecast screens, except the last one, handle the conversion from nt_xxxxx already.  My question still is, why is the last entry different.

Hopefully I'll have some time this week to track down the problem.

lorol, 2017-09-18:
Basically there are 2 issues. One is due to overlapping identical keys when scanning the forecast json, the simplified section with manipulated index overrides.

The second issue is missing night time icon's byte arrays. It shows once you try to fix 1) so you either have to manipulate the string by cutting nt_ (then points to daily) which will show you sunny picture for the night, or better add the night time icons.    
Daniel, Is there a software to create a compatible 2-bits depth arrays from, say BMP?

herrold, 2017-09-19:
@lorol asked
... Is there software to create a compatible 2-bits depth arrays from, say BMP

The FOSS offerings of: 
- ImageMagick (has the 'convert' command line tool)
-  The Gimp (full blown graphical suite)

come to mind.  I have used 'convert' for years to make 'older' formats such as .xpm and so forth

G6EJD, 2017-09-19:
LCD image converter can do that: http://www.riuson.com/lcd-image-converter
But the final file needs to be converted to Xbox format and there are a few online converters for that part.

lorol, 2017-09-19:
Hi, I tried ImageMagick 'convert tool. While it is very reach of features, nothing generates even close to the format used by the project. Standard XBM format doesn't  define 2-bit depth.  Given source bmp can be converted to XPM or 2-bit depth png, by `convert chanceflurries.bmp [-negate] -colors 4 -depth 2 test.[xbm|png]`  but still far from ready to be used.

@G6EJD can you please elaborate further regarding this "Xbox" format?

BoxOfSnoo, 2017-09-22:
Thanks @lorol - do you have some code for that that we can use?

lorol, 2017-09-22:
Yes here I posted a link to my tool: https://support.squix.org/133/tools-for-making-bitmaps-for-wx-color
If you need, I can add later my project .cpp and .ino files changes I made to get this working.

lorol, 2017-09-23:
[added_nt_and_last entry_fix.zip](https://github.com/squix78/esp8266-weather-station-color/files/1326270/added_nt_and_last.entry_fix.zip)

Here are all changed files.

BoxOfSnoo, 2017-09-23:
Thanks!  You should create a pull request so it can be merged into the main code, too.

lorol, 2017-09-25:
I prefer to leave @squix78 to consider and implement my changes with no messing with code tree.
BTW, if any of my 4 added night pictures (2 big + 2 small) of the zip show scrambled on the screen, please regenerate them or yours by the [tool](http://web.295.ca/295forums/phpBB2/display.htm) I just updated.

lorol, 2017-09-26:
[Here](https://github.com/squix78/esp8266-weather-station-color/files/1333601/added_nt_and_last.entry_fix_updated.zip) are all changed files and added night time icons, I used to fix the issue.

BoxOfSnoo, 2017-09-28:
A fork with a pull request would still probably be the best way to get the code to @squix78 (and everyone else).  Merging in future changes would be a lot easier too.  If nothing else, we could just use your fork!

I appreciate the fixes you did.

lorol, 2017-09-28:
Thanks for the feedback @BoxOfSnoo. 

I just updated (again :) my [zip](https://github.com/squix78/esp8266-weather-station-color/files/1340641/added_nt_and_last.entry_fix_updated_Sep_28_2017.zip) file attached also above to fix one typo. 
Ok I'll do a pull request. I may be possible a fix merge of the lib with backwards compatibility to avoid breaking of the old projects that share it.
 

I was playing with my[ JS picture tool](http://web.295.ca/295forums/phpBB2/display.htm) challenge. It is very messy but now it reduces any picture down to four chosen colors and makes a compatible c code.  
I found that the picture width rounding should be to 4 not to 8 points. It looks like there is a bug of the original squix78's arrays - they have all extra bytes of 8-point rounding at the end as unnecessary 0x00's appended. Anyway ... hope someone would build and share new icon graphics set for fun.  BTW, I use Mozilla browser, on others it may not work.

pilnikov, 2018-04-29:
I want to add a little ....
if there is no information in the prognosis, the name of the icon remains empty (""), but for draw on the screen name of the icon must not be empty. For correct work, I propose to initialize the "Icon" field as "unknown".
For example in WundergroundConditions.h
` String weatherIcon = "unknown";`

G6EJD, 2018-04-29:
I've notice that the WU Servers have been returning odd values in the last 24-hours (not unusual), so adding a check for unknown values if worthwhile doing.

marcelstoer, 2019-02-19:
[Weather Underground closed their existing API](https://apicommunity.wunderground.com/weatherapi/topics/end-of-service-for-the-weather-underground-api) for non-paying customers on 2018-12-31. After [they stopped issuing free API keys](https://thingpulse.com/hello-openweathermap-bye-bye-wunderground/) for low-volume personal use in May we migrated to OpenWeatherMap and [deprecated all WU code](https://github.com/ThingPulse/esp8266-weather-station#deprecation-notes) in this library.

We are now closing all WU-related issues and PRs.

closed 2019-02-19 by marcelstoer