ThingPulse/esp8266-weather-station-color #41: Generating icons

Luc3as, 2017-09-09:
Hello, 
I just want to ask if there is some way for generating images and icons to progmem palleted arrays, so it will be correctly displayed by minigrafx library. 

thanks

G6EJD, 2017-09-09:
You can take an image and generate the .h files using:

http://www.riuson.com/en/lcd-image-converter/download

The current images are already located in PROGMEM.
To add new images create the .h and paste in making sure your have the correct height and width dimensions set, etc. You need to get the correct format. I'm not sure if that has been published?

Luc3as, 2017-09-10:
I spent whole evening playing with this software but did not find any suitable settings to generate image right. when I try to substitute image in PROGMEM with bits generated from software all I get is just smashed up pixels not actual image. Is there some special settings like directions of slicing the image in software or templates you could provide ? 
many thanks

G6EJD, 2017-09-10:
Let's say you want to display a new icon, or a different one, find you image first, get it to  the correct size and format, which needs to be 
Bit-Depth: 2 // Monochrome
Width: 99
Height: 100

Next start the programme, create a new image, then Image - Import Image,  your new image. 

Now go to Options, Conversion, then on the top line select 'Monochrome', then 'Preview

You will get many lines looking like this: 0xff, 0xff, 0x07, 0xff, 0xff, 0xf8, 0x00, 0x7f, 

Go to the 'weathericons.h' file,  probably go to the end and enter:

const char MyNewImage[] PROGMEM = {
  0x01, // Version: 1
  0x02, // BitDepth: 2
  0x00, 0x63, // Width: 99
  0x00, 0x64, // Height: 100
0xff, 0xff,...all 99x100 bytes so quite a lot...
};
then paste you new data in after height

Save it, then you can use the new image. You can generally make out if your new image is correct if you align the number of bytes across the page in (in this format) 99/8 bytes across= 13bytes across, so you'll have 96 rows of 13 bytes. But note you have to round up the byte count as Daniel says, this is because 99-bits or 100-bits does not fit into an exact number of bytes. I usually make all mine in this example: 96x96 or 96x104
Hope this helps.

Luc3as, 2017-09-10:
definitelly I am doing something wrong,
I think it is in generating or converting the source image. 
I understand principle of how it is stored to progmem and how it is pixel by pixel drawed on display, but best I can get from it is this
[image]
I am trying it with logo because its easiest to see every reboot. I tried to generate BMP or PNG or even XBP file , and then convert it to monochrome profile by LCD image converter, strange thing is that is is always bigger , for example in original logo there is 2800 hex numbers, when I convert image with same dimensions I get more than 10 000 numbers. 
I tried to create monochromatic image from photoshop, gimp, even ms paint, or convert it online through  some websites. 
can you advise  how to get correct format of image ? 
many thanks for your responses

G6EJD, 2017-09-10:
I've' been here! :) it is not easy when the information is sparse and reverse engineering is needed. First of all, it looks like you may not have the icon size correctly defined. did you carefully (this is how I did it) make an icon of say 25x25 pixels using mspaint would do and draw a line vertical and horizontal. Then go through the process above, you should be able to see the vertical line as a column of 0xff and same for the horizontal line 0xff (or inverse maybe so 0x00 for a line, etc). Then preview and copy into the library, making sure all the parameters are defined to match the image, did you do all this for you icon/image?

Daniel uses the XBM file format for his SSD1306 driver, so there's a possibility he used this format for this library, I have not checked.

G6EJD, 2017-09-10:
Which function are your calling?
void drawXbm(int16_t x, int16_t y, int16_t width, int16_t height, const char *xbm); 
void drawBmpFromFile(String filename, uint8_t x, uint16_t y); 
void drawBmpFromPgm(const char *xbm, uint8_t x, uint16_t y); 

One of them is XBM formatted files.

G6EJD, 2017-09-10:
https://blog.squix.org/2015/05/esp8266-nodemcu-how-to-create-xbm.html

Luc3as, 2017-09-13:
You are the man 🥇 
with changing function to xmb I was able to replace the image and display it correctly.
would be nice to get some tutorial on how to generate images to progmem the Daniel's way, but for now I am happy with results as it is. 

Once again big thank you ;)

closed 2017-09-13 by Luc3as

G6EJD, 2018-05-07:
Your process looks OK and your doing exactly what I do, try repeating it but with a BMP version of the image. PNG are compressed images. As you have done I tend to stick to integer byte size images e.g. 8x13=104x104 then I can tell if the image data is correct by a symmetrical number of row/column bytes. In your case there should be 13x 13 rows.

I'm guessing you are getting some form of display up, just garbled?

marcelstoer, 2018-05-07:
Yeah, I think we should document this on https://docs.thingpulse.com/ ("How-tos") at some point. That is basically just another public GitHub repository. So, if anyone is willing to contribute providing PRs is often faster than waiting for us to shift our priorities 😉

klimbot, 2018-05-10:
Thanks for the suggestions, I've worked out the image display problem now, I hadn't understood the palette/bitdepth part of the library and wasn't accounting for it properly. I was outputting in monchrome but I have four colours in my palette with the example sketch I started with.
`uint16_t palette[] = { ILI9341_BLACK, ILI9341_WHITE, ILI9341_YELLOW, ILI9341_BLUE}; `

I've changed my conversion settings to "Grayscale 4" and now I am at least seeing the right image displayed but the colours are all out of whack. Any idea what I need to do to fix it? Colours that are supposed to be white (in the BMP) are showing up as blue, black is black and the others just look strange.

lorol, 2018-05-14:
See this if you like to play with 4 colors.
It is JS, so no server side, you can run it on your PC offline.
It is a bit old, some tweaks on variable types in header may be needed:

https://rawgit.com/lorol/esp8266-weather-station-color/master/resources/converter-tool/display.htm

klimbot, 2018-05-20:
@lorol thanks for sharing the website, worked like a charm! 
Headers were correct most of the time but on one occasion it put out the wrong width/height. 

Thanks again

marcelstoer, 2018-05-20:
Great! @klimbot or @lorol would you be willing to contribute a short list with instructions for our documentation?

lorol, 2018-05-20:
No problem @marcelstoer . I shared long time ago this one for 4-color pictures and one for font editing. The only problem is that these nice ThingPulse projects went to crazy forking and splitting and people are lost. Appreciate that you are trying to put all this in order! Just let me know how do you envision better exposure of these "tools" and we will move and explain them. Feel free to grab them from my GitHub.

pilnikov, 2018-05-22:
is it possible to download icon files by esp directly from the github and not from the squx blog?

lorol, 2018-05-22:
If you mean to get SVN-updated "live" icons from here through Internet, there are limitations:
- Live bitmaps data, even for small 4-color cloud/sun is "huge" comparing to Json's - ESP would struggle. That's why they are stored to ESP
- GitHub servers do not like much continuously automated raw downloads by multiple clients from Internet.