ThingPulse/esp8266-weather-station-color #42: Timezone

soubs, 2017-09-12:
Hi There,

Just wanting to find out if India timezone can work on this device. That is UTC +5.30 hrs. I did a bit research across posts, it seems like the +30 mins is not supported by the library. Has there been any update to this. 
This is a really interesting project and would like to gift the project to a friend from India.
Thanks

soubs, 2017-09-19:
No one got any views ??

soubs, 2017-09-21:
Thank you so much for getting back. I tried it, still can't get the 0.3 mins working. The time is either 30 mins ahead or 30 mins behind

G6EJD, 2017-09-21:
It’s because the hour variable is a integer and so it’s being rounded up or down to the nearest hour, that variable type would need to be changed throughout the programme to enable 0.5 hours to be held and hence displayed correctly.

G6EJD, 2017-09-21:
I had a look at the source code, only in 12-hour mode is the hours an integer, line 318.

What did you use for configTime(UTC_OFFSET * 3600, 0, NTP_SERVERS);

UTC_OFFSET is default 1, try
#define UTC_OFFSET + 1.5

And  set the clock to 24-hour mode.

herrold, 2017-09-22:
@soubs  I was thinking about this, and perhaps scaling the 24 hours in a day to 240 segments (so 10x as many) is a way forward -- it will still fit in an 8 bit CHAR, and permit using an offset of 5 rather than .5 (a FLOAT)

This would be at the expense of more code to do the scaling, of course

G6EJD, 2017-09-22:
The first test I would do is add 1800 to the *3600 part of the configtime statement so that it is *5400 as that adds 30-mins and see how this affects the time display. Then change the variable type to float and try again, so small steps. It’s only around line 318 where the code needs to be changed.

G6EJD, 2017-09-22:
It’s easier than, as a quick trial at line 257 configTime(UTC_OFFSET * 3600, 0, NTP_SERVERS);
change it to configTime(UTC_OFFSET *1.5 * 3600, 0, NTP_SERVERS);
Or if your 4.5 hours ahead then *4.5 
Or just change the Utc offset in the config file to 4.5 currently 1

G6EJD, 2017-09-22:
In settings.h on line 54 #define UTC_OFFSET + 1 change to #define UTC_OFFSET + 5.5 for India

soubs, 2017-09-23:
@G6EJD Thank you so much for your advice, it worked. Below is the code that I used in the setting.h as you mentioned.

// Settings for India
#define UTC_OFFSET +5.5
struct dstRule StartRule = {"IST", Second, Sun, Mar, 2, 1800};
struct dstRule EndRule = {"IST", First, Sun, Nov, 1, 1800};  

Can't take the DST rules out as there are dependencies so added the same rule for both the changes. Also, the 12 hr seems to be working. I will use it for a couple of days a see how it goes. Will report back in case of any issues

G6EJD, 2017-09-23:
Those changes will have solved the India Time Zone and no code changes which is good. You should have no further problems now. Glad it's worked out.

soubs, 2017-09-23:
I think I have jumped the gun here .. All seems to be working fine, however the time reverts to 12:00:00 after the first update and not updating the time or weather.

G6EJD, 2017-09-23:
This is most likely due to your internet connection, or choice of time server, perhaps use http://www.pool.ntp.org/zone/in to use a more localised server.
Did it ever show the correct time?

55 struct dstRule StartRule = {"CEST", Last, Sun, Mar, 1, 3600}; // Central India Summer Time = UTC/GMT +2 hours 
56 struct dstRule EndRule = {"CET", Last, Sun, Oct, 1, 0};       // Central India Time = UTC/GMT +1 hour 

Does your DST move by only 30-mins? You have used 1800=30-mins?

Time server:
#define NTP_SERVERS "0.ch.pool.ntp.org", "1.ch.pool.ntp.org", "2.ch.pool.ntp.org"

should be:

#define NTP_SERVERS "0.in.pool.ntp.org", "1.in.pool.ntp.org", "2.in.pool.ntp.org"

OR:

#define NTP_SERVERS "0.asia.pool.ntp.org", "1.asia.pool.ntp.org", "2.asia.pool.ntp.org"

soubs, 2017-09-23:
I have already made one for myself in Sydney which is +10 UTC and it works flawless.

I have added 1800 to adjust the time to IST. 

I have updated the NTP servers as you mentioned however still getting the same result.

G6EJD, 2017-09-23:
But does time (mins) move by 30-mins ?

My npt clock increments time supplied to the config time statement in 30-min steps and that works just fine. Daniels software then displays time from the server and there is no manipulation of the variables try putting some diagnostic print statements around the config time function or monitoring the serial port to see if there are any runtime errors.

G6EJD, 2017-09-23:
Alas, it seems there is no answer to this problem as I have now discovered the time library ignores the DST parameters:
configTime(timezone, DST, NTP_SERVERS);
Normally:
configTime(5.5 * 3600, 1800, NTP_SERVERS); for India it seems.

So the time library will accept non-parameters but does not act on them, effectively only using 5 for 5.5 offset and furthermore it completely ignores the DST element!

There is forum discussion and folks have done what I have and produced a small test programme to confirm 'configureTime' is not working as expected.

See here: https://github.com/esp8266/Arduino/issues/2505

soubs, 2017-09-24:
Hi there,

I think there is some hope.. I think I had a bit of issue with my wifi , after changing it , it's running for 3 hrs at the moment and it's updating. 

Fingers crossed.. will update after running it for a few days now.

G6EJD, 2017-09-24:
Good to hear, but the configtime library function can’t handle 30-min offsets which is a surprise to me as I trialled it like others to confirm the function is defective so the best you will get is 5 or 6 hours offset. Clearly the fact that some parts of the world that have 1/2 hour offset increments was not considered during its build. I have not tried other time libraries as these may well have solved this issue.

soubs, 2017-09-28:
Hi @G6EJD ,

Update, the timezone seems to be working fine. What I realised is that the code now assumes the time is by default 12:30:00 so it works. But now we got a different problem - Since there is no DST rule in the India timezone it works with the start rule but goes all wrong with the end rule. Any suggestions, we are this close to getting it done !! And I can't seem to ignore the DST rules.

#define UTC_OFFSET +5.5
struct dstRule StartRule = {"IST", Second, Sun, Mar, 2, 1800};
struct dstRule EndRule = {"IST", Second, Sun, Nov, 2, 1800};

G6EJD, 2017-09-28:
How are you moving time forward? The NTP and time library use UNIX time so you’d need to add an offset to ordinal time to do it properly.

soubs, 2017-10-11:
Apologies for the late reply, have been travelling for work. I can get the code to work , but the dst rule ruins it all and I can't fix that. Any help will be most appreciated

G6EJD, 2017-10-11:
The library being used for DST does not support 30-min increments, you could ask the author to modify it, really it's an issue on Github because there are countries in the world like yours that have 30-mins offsets.  It would have been better if the library had been designed with that constrained at the onset.

soubs, 2017-10-12:
Hello , 
Am back, maybe with good news. 
So, the issue was, I was being able to get that timezone working but the DST rule was playing up. I never thought about the DST rules since where I live I need it, I was trying to disable the DST rule but the code wont allow it, so my solution is the repeat the DST start rule in a loop cause the dst end rule messes up the time. 

Code fix that I used:

settings.h:
// Settings for India
 #define UTC_OFFSET +5
 struct dstRule StartRule = {"", Second, Sun, Oct, 12, 1800}; 
 //struct dstRule EndRule = {"IST", First, Sun, Nov, 1, 5400};

the main code file:
// Setup simpleDSTadjust Library rules
simpleDSTadjust dstAdjusted(StartRule, StartRule);  // Settings for India
//simpleDSTadjust dstAdjusted(StartRule, EndRule);

To me logically it should work.

Any insights, most appreciated.

Thanks
S

G6EJD, 2017-10-12:
If your using a whole number DST offset it should work OK but using 0.5 hour increments don’t. Check out the DST library details as that’s what determines success or not.

marcelstoer, 2018-04-15:
I see little benefit of keeping this open any longer. Everything there is to be said has been said.

The relevant external issues are as follows:
- https://github.com/esp8266/Arduino/issues/2505
- https://github.com/neptune2/simpleDSTadjust/issues/2#issuecomment-298045845

closed 2018-04-15 by marcelstoer

Priya-Yadav-99, 2020-08-11:
> Hi There,
>
> Just wanting to find out if India timezone can work on this device. That is UTC +5.30 hrs. I did a bit research across posts, it seems like the +30 mins is not supported by the library. [...]

Hey!
I read out your query and also suffer from this. I would like to ask which compiler and libraries have you used to run this code and get time in IST +5:30 in ESp8266.

Thank you