ThingPulse/esp8266-weather-station-color #46: Time shows 00:00:00 for short time after forecast update

lorol, 2017-09-30:
configTime(UTC_OFFSET * 3600, 0, NTP_SERVERS);  can take several seconds to update the system clock.
If time / date is displayed during this period, it will be zero-ed.

Suggestion, in  **drawTime()** prevent the display unless the date is correct

` if (timeinfo->tm_year > 116) { // valid, if the year is > 2016`

See [here](https://github.com/lorol/esp8266-weather-station-color/commit/82863e760c04c2036d74209f53b4e2f395a6d409)

safari2, 2017-10-31:
The issue is not solved in current release. "Fix time synchronization (1970 00:00:00) issue" works not always.
[Proof ](url= https://www.dropbox.com/s/9g2a3axps8e19xy/colorWeheatherTimeBug.jpg?dl=0 )(new sketch was uploaded yesterday)

lorol, 2017-10-31:
Hi @safari2 - which sketch do you have in your reference picture? 
The main repository was not updated several months.
If you use my workaround recommendation as I showed in link, you will just not see the clock (black space there) for several seconds until synchronizes. 
At least the annoying zeroes will not show. 

The root cause is the network delay response from NTP servers. You can also chose other (closer to you, less busy, faster responsive) NTP servers in your setting.

safari2, 2017-10-31:
this one:
https://github.com/squix78/esp8266-weather-station-color
there are comments "Fix time synchronization (1970 00:00:00) issue" 3 month ago.
For me this solution works good (like your but in cycle):
while (timeinfo->tm_year<117)
  {
   configTime(UTC_OFFSET * 3600, 0, NTP_SERVERS);
   timeinfo = localtime (&now);
   Serial.println(timeinfo->tm_year);
   delay(500);
  }

lorol, 2017-10-31:
Yes, it is OK, but the problem is if it takes too long ... or the NTPs
connectivity is unavailable for long periods, then the main process will
loop here?
Also I am not very sure what configTime triggers on background (process
could be long), and should you re-attempt all this in 500ms frequency.
I think configTime behind the scenes is doing multiple attempts with no
need to loop - please check in depth.

safari2, 2017-10-31:
My version worked good more than half of year. The second request to NTP-server always works. If I understand correctly that the library provides for rotation of NTP-server's IP-addresses.
There was no problem, but in my country no more summer/winter time. 
And this week I found that my weather station was switched to winter time. I had to fix in the sketch. At the same time decided got on the site to see what's new, saw the update. I've decided that the problem 1970 is solved. 
The result was is so-so :)

lorol, 2017-10-31:
All good ... IMO I just would be cautious to add extra while loops with no guaranteed escaping in main loop.

closed 2018-01-20 by lorol