ThingPulse/esp8266-weather-station-color #48: Carousel not smooth

dkotte, 2017-11-23:
First i build weather station with with small OLED display. Nice.
Now i build weather-station color. Great. Thanks a lot!
But the carousel from weatherstation color change pages with jumps and is choppy while at the small device it runs very smooth. That gives problems with the WAF :-) 
Is there an error i made or an option to optimize the page-change?
Where i can set the time used for page change?

rickylee64, 2017-11-29:
same issue for me. the display cycles through 5 pages way too fast, about 5 pages in 2 or 3 seconds. a touch screen issue maybe?

marcelstoer, 2018-04-14:
This is actually a dup of #22. @stamatoulis thanks for the workaround.

closed 2018-04-14 by marcelstoer

dkotte, 2018-04-15:
Hi marcelstoer,
the problem here is NOT the cycling from one infoscreen to the other because the missing touch chip/connection. (a similar workaround for this problem i build too)
My problem here is the display quality within the carussel function.
PS: auf deutsch könnte ich das Problem genauer beschreiben.

MarioP73, 2018-04-15:
Schreib mal auf deutsch