ThingPulse/esp8266-weather-station-color #49: touch calibration

romain3w, 2017-12-05:
I have some issue with the calibration of the touch screen. it appear that x and y are reverse. hardware or software issue ?

G6EJD, 2017-12-06:
Have you rotated your screen? Normal orientation is portrait, if so swap you touch x and y around

romain3w, 2017-12-06:
Screen is not rotated. I took the .ino and only change the wifi parameters !

romain3w, 2017-12-06:
Can be fix replacing in TouchControllerWS.cpp :
int x = (p.y - ax) * dx;
int y = 320- (p.x - ay) * dy;

BY :
int x = 240 - (p.y - ax) * dx;
int y = (p.x - ay) * dy;

WrQth, 2017-12-29:
I'm having trouble with touchscreen calibration as well though I have rotated the screen with gfx.setRotation(1). Is there something in TouchControllerWS.cpp that needs to change match the new orientation?