ThingPulse/esp8266-weather-station-color #53: configportal.h:183: error: assignment of read-only location

Kabron287, 2018-01-03:
Maybe I and Arduino IDE do not understand something in C, but how can you comment such assignments, previously #defined as constants?
WIFI_SSID = value;
WIFI_PASS = value;
WUNDERGROUND_CITY = value;
etc.

G6EJD, 2018-01-03:
// WIFI_SSID = value;
// WIFI_PASS = value;
// WUNDERGROUND_CITY = value;

Kabron287, 2018-01-03:
Well done! What about:
WUNDERGROUND_COUNTRY = server.arg("wundergroundstate");
WUNDERGRROUND_LANGUAGE = server.arg("wundergroundlanguage");
WUNDERGRROUND_API_KEY = server.arg("wundergroundkey");
SAVER_INTERVAL_SECS = server.arg("timeSaver").toInt() * 60;
SLEEP_INTERVAL_SECS = server.arg("timeSleep").toInt() * 60;

configportal.h:325: error: 'CONFIG_SSID' was not declared in this scope

etc...

G6EJD, 2018-01-03:
You can comment all those out to preserve the definitions in case you use them or:
WUNDERGROUND_COUNTRY = UK
 WUNDERGRROUND_LANGUAGE = EN
 WUNDERGRROUND_API_KEY = your key needed
 SAVER_INTERVAL_SECS = 60 // for 60-secs
 SLEEP_INTERVAL_SECS =  3600 // for 1-hour

I think there is a separate file that holds your SSID information

closed 2018-04-14 by marcelstoer