ThingPulse/esp8266-weather-station-color #55: resets on usb loss

kolasinski, 2018-02-14:
I'm not sure if this is an "issue" or that I do not understand something. 

When uploading code via Arduino IDE, weather station Color works fine, connects to my network and pulls down data, displays great. Works great.

I am only connecting via micro USB - no battery. When I unplug, and replug, unit loses memory and requires re-uploading the program via Arduino IDE.   

Do I need to always keep a li-po cell on the unit? Or is there a setting that I just do not see to keep the settings after power down?

Thanks

al-ang, 2018-03-26:
Are you saying the device needs to be re-programmed (flashed) every time the device is booted?  If so, then there's something very wrong. Also, if that's the case, the help on the Arduino forums might be more helpful. Hopefully, this isn't the case.

G6EJD, 2018-03-26:
There may be an esp.reset() in the programme leading to loss of WiFi config data that then gets cleared at restart. I haven’t checked the code.

pilnikov, 2018-03-26:
Please add 

So now

`void connectWifi() {
  if (WiFi.status() == WL_CONNECTED) return;`

It is necessary to add

`void connectWifi() {
  WiFi.disconnect(); //added
  WiFi.mode( WIFI_STA ); //added
  if (WiFi.status() == WL_CONNECTED) return;`

G6EJD, 2018-03-26:
Agreed

marcelstoer, 2018-04-14:
@pilnikov could you please reference a source that explains why it should be necessary to add those two statements?

marcelstoer, 2018-04-23:
```c
WiFi.disconnect(); //added
WiFi.mode( WIFI_STA ); //added
if (WiFi.status() == WL_CONNECTED) return;
```

How likely is it that the conditional on line 3 ever returns `true` if you do a `WiFi.disconnect()` right before...

pilnikov, 2018-04-23:
After disconnect we true connect with WL_CONNECTED. WiFi.disconnect() resets the settings of previous connection.

pilnikov, 2018-04-23:
Check your devboard. If have a capacitor on the ground in the reset chain - then you will hardly be able to repeat the described situation.

marcelstoer, 2018-04-23:
Still doesn't make sense to me. I know what `WiFi.disconnect` does and, therefore, the described fix is illogical. After disconnect `WiFi.status()` will of course return 6 (`WL_DISCONNECTED`). Hence, the conditional will always evaluate to `false`.

> Check your devboard.

No devboard here. I'm testing against our ThingPulse reference hardware. 

@G6EJD & @al-ang you both seem to be in favor of the proposed fix. Why would you want to explicitly disconnect from the AP every time first? Known issue with the SDK I'm not aware of? Furthermore, how would that be related to the OP's issues which is described as "When I unplug, and replug, unit loses memory"?

G6EJD, 2018-04-23:
Marcel, your correct. What I was responding to (I thought) was reference to a problem I’ve seen many times when the ESP8266 comes out of sleep and fails to reconnect the WiFi but adding the WiFi.disconnect and WiFi.mode(WiFi _sta) does then result in reliable reconnection. I’m using the Arduino IDE and this may well be a facet of the IDE.

I think @kolasinski may well have misconstrued a connection problem or some other problem with a credentials memory clearing event but never got round to checking for an esp.reset() in the code. Also it’s been a long time since he reported the issue.

pilnikov, 2018-04-24:
I also agree with @G6EJD, @kolasinski may well have misconstrued, the ESP may "fall into a stupor" (WiFi part), return it only WiFi.disconnect().

pilnikov, 2018-04-24:
https://github.com/esp8266/Arduino/issues/2174

closed 2018-04-24 by marcelstoer