ThingPulse/esp8266-weather-station-color #58: Screen Rotation

556duckvader, 2018-03-14:
How do you go about rotating the screen to landscape?  I have looked and I am not really seeing where to configure that.

G6EJD, 2018-03-14:
Well the simple adjustment would be to swap screen height and width, but it's not as simple as that as the minigfx library has no screen rotate and the whole project has been designed (x,y layouts) to be 240x320. You could write a function to rotate the screen and go through the whole programme to adjust all the x and y offset values for landscape.

Has anyone done this?

pilnikov, 2018-03-15:
[image]

marcelstoer, 2018-04-15:
@G6EJD actually answered this -> closing

closed 2018-04-15 by marcelstoer