ThingPulse/esp8266-weather-station-color #60: multiple wu calls can be combined

al-ang, 2018-03-24:
It looks like every WU update has at least 3 WU API calls (conditions, forecast, astronomy).  These 3 calls can be combined into 1 WU API call by combining them into one request.  e.g.

http://api.wunderground.com/api/<api key>/astronomy/conditions/forecast/q/CA/San_Francisco.json

Combining calls would allow the update to be called more often and remain under the free api call limit meaning the displays would have more accurate dat.  I combined calls by modding an old version, but The current update looks much more complicated and has thrown me for a loop.  Would it be possible to combine WU API calls with the latest design structure (please)?

G6EJD, 2018-03-24:
Have you confirmed this? My understanding is each request for a data set counts as a call so combined or separate amounts to the same API usage count.

al-ang, 2018-03-24:
Yes, all the data can be accessed with one api request.

G6EJD, 2018-03-25:
You’ve made an excellent discovery/observation and whether it can be implemented on the ESP devices will come down to the way the data is received and it’s volume, about 15KB of data. Making three calls is a nice clean and easy to implement approach, conversely the single call needs careful detection and management of the section separators and currently this code is not structured to do that - Who’s Going to have a go at rewriting it?

Conversely weather does not change that quickly and a 30-min update frequency keeps the api calls well within the 500 allowed per day, so does it matter if the api call rate is (24*2*3) 144 or (24*2) 48 probably not, but it’s good to know we can do this.

al-ang, 2018-03-25:
Daniels's json parser handles the single call well.  I hacked one of the original implementations (a version after fowler's changes) and added an updateALL -- it worked perfectly.  

My current problem came when I looked at the current codebase -- the separation of call content seems to have gotten stronger and it looks like content structures have been added.  I started to make my changes, but got lost in the myriad of other necessary changes and did not finish.

It's my hope that someone with a better mental model of the code can add the single call feature at a place where it's most efficent.  I really like being able to refresh the data more frequently.

There are some features of the new codebase I really want to get into and play with like the faster screen refresh and the carousel.

marcelstoer, 2018-04-14:
@al-ang that's an interesting observation, thanks. Just like @G6EJD I personally see little benefit in more frequent updates (WU quota thus not an issue) but for battery powered devices with deep-sleep, see https://github.com/ThingPulse/espaper-weatherstation, it makes a huge difference  whether you make 1 call or 3.

Hint: https://thingpulse.com/2018/03/12/5-hacks-prolong-espapers-battery-run-time/

al-ang, 2018-04-18:
I hadn't thought of power saving possibilities.  That's a new view for me.  I like how GitHub projects provide new ways of looking at things.

 I had been trying to increase the number of updates to keep data from being stale in rapidly changing situations but still stay within the number of api calls in the free plan.

G6EJD, 2018-04-18:
In my latest project epaper weather station I use a single (multiple request) call asking for conditions, forecast and lunar data and it all works so thanks for the idea, however the amount of data returned is relatively large at ~12K and on the ESP32 it does take a relatively long time to receive and decode it, much longer than 3 separate calls but I can’t explain why other than the ram usage. I tried to do this with the ESP8266 but there is no where near enough memory resources to run the code whereas it can be done with 3 separate calls easily. I don’t use the Thingpulse streaming JSON decoder so it might work with that. I had a look at recoding the espaper weather station code to use a single call and at first I thought it would be very straightforward but gave up when I realised the extent of the interdependencies that needed to be changed.

marcelstoer, 2018-04-18:
Whoever might eventually take a stab at this should consider that it has to be implemented in the weather station lib at https://github.com/ThingPulse/esp8266-weather-station/tree/master/src (i.e. not _this_ repo).

squix78, 2018-04-18:
Exactly because of RAM usage I split up WundergroundClient into separate clients, so that the memory could be allocated only for the methods of interest. Especially in the color kit where we use part of the memory for the screen buffer RAM is extremely scarce. There's not even enough memory for a SSL connection, if I remember correctly.

G6EJD, 2018-04-24:
Out of interest, I have dropped using one call for three API responses /conditions/forecast/astronomy because it's not reliable and today the responses were too big (as weather conditions vary) for the ESP32 to handle using the standard Arduino JSON, so I shall have to transfer my code to Daniel's streaming variant. For now I call WU twice /conditions/astronomy then /forecast

G6EJD, 2018-04-24:
My memory usage (3 combined) was ranging from ~8K to 15K and today my reader was locking all the time, after some diagnostic I discovered I was out of memory, but previously it was working fine, I have not studied the WU API responses, but I Note the moon phase changed and their weather forecast text was very long about 3 -lines of text on the 4.2" e-paper, so response size does vary. When I separated the calls back to two it worked OK again. I can still keep well within the WU 500/day limit with my three (test) stations receiving API data, I was averaging about 150/day, so that will increase.

G6EJD, 2018-04-25:
@al-ang Why not share your solution for a single API request as a PR and make it an alternative version, it’s a team effort and part of the Thingpulse ethos to share.

marcelstoer, 2018-06-16:
WU is deprecated: https://github.com/ThingPulse/esp8266-weather-station#deprecation-notes

closed 2018-06-16 by marcelstoer