ThingPulse/esp8266-weather-station-color #61: Cyrillic font

pilnikov, 2018-03-25:
Hello. Is it possible to add Cyrillic characters at position 192-255 in standart font?

lorol, 2018-03-26:
Yes, it is possible. but someone needs to do it and merge to main code. Not so many Cyrillic people :) here.
See for idea my way: https://github.com/lorol/esp8266-oled-ssd1306

There are 2 things modified
1) Implement case 0xD0: and  case 0xD1: to 
uint8_t OLEDDisplay::utf8ascii(byte ascii)

2) A Cyrillic font bitmap code

pilnikov, 2018-03-26:
Спасибо

pilnikov, 2018-04-10:
Lorol Thanks you for the support! You guided me along the right path.
But in the "process" there are new challenges.
The problem, it seems to me, is hidden in the MiniGrafx library. It is fork of adafruit gfx.
I can not display characters with the number > 127
I tried printed string like as "\xA2\xA3\xA4" but the result is "empty"
Although the string "\x7D\x7E\x7F" produces "}~ "
Maybe you already solved this problem?

G6EJD, 2018-04-10:
You need to you the squix provided font converter and use a font that had characters above 127 nearly all of the examples are blank above 127 and that’s your problem. You cN covert fonts to the squiux or adafruit fornat using a Unix tool provided by adafruit it they give an example of using it in a windows environment. Do you have a font with the required characters in it?

pilnikov, 2018-04-10:
[image]
Yes, of course, I'm using a font converter for specifically for Squix. And I have the characters after 127
`  0x0E, // Width: 14
  0x11, // Height: 17
  0x20, // First Char: 32 
  0xE0, // Numbers of Chars: 224

........................................ // Part of Jump Table:
  0xFF,0xFF,0x00,0x06, // 158
  0xFF,0xFF,0x00,0x06, // 159
  0xFF,0xFF,0x00,0x06, // 160
  0x08,0xE0,0x0C,0x07, // 161
  0x08,0xEC,0x1A,0x0B, // 162
  0x09,0x06,0x1A,0x0B, // 163
  0x09,0x20,0x17,0x0B, // 164

............................................... //Part of Font Data:
  0x00,0x00,0x00,0x08,0x00,0x02,0x08,0x00,0x02,0xF0,0xFB,0x01,0x00,0x04,0x00,0x00,0x04, // 125
  0x00,0x00,0x00,0x00,0x02,0x00,0x00,0x01,0x00,0x00,0x01,0x00,0x00,0x01,0x00,0x00,0x02,0x00,0x00,0x02,0x00,0x00,0x02,0x00,0x00,0x01, // 126
  0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x40,0xFF,0x03, // 161
  0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x1F,0x00,0x80,0x20,0x03,0x40,0xF0,0x00,0x40,0x4E,0x00,0xC0,0x41,0x00,0xB8,0x20,0x00,0x00,0x11, // 162
  0x00,0x00,0x00,0x00,0x41,0x00,0xE0,0x31,0x00,0x10,0x2F,0x00,0x08,0x21,0x00,0x08,0x21,0x00,0x08,0x40,0x00,0x10,0x40,0x00,0x20,0x20, // 163
  0x00,0x00,0x00,0x00,0x00,0x00,0x40,0x0B,0x00,0x80,0x04,0x00,0x40,0x08,0x00,0x40,0x08,0x00,0x80,0x04,0x00,0x40,0x0B, // 164
`

G6EJD, 2018-04-10:
OK, so if I do drawString(0,0,"°"); I get character 176 if you type the ASCII code sequence in I types ALT 176 to get the degree symbol, do you see your characters from your font. Do you see the same for e.g. 'A' ALT 065

lorol, 2018-04-10:
Hi @pilnikov you dont need to use x based strings. This is the idea. 
1) Just use the Cyrillic text with Arduino IDE in quotes: "Глав"
2) Yes, in MiniGrafx.cpp you need to implement case 0xD0: and case 0xD1:  in utf8ascii function same way as I showed earlier.

pilnikov, 2018-04-11:
Try this function with symbols> 127
`// Code form http://playground.arduino.cc/Main/Utf8ascii
uint8_t MiniGrafx::utf8ascii(byte ascii) {
  static uint8_t LASTCHAR;

  if ( ascii < 128 ) { // Standard ASCII-set 0..0x7F handling
    LASTCHAR = 0;
    return ascii;
  }

  uint8_t last = LASTCHAR;   // get last char
  LASTCHAR = ascii;

  switch (last) {    // conversion depnding on first UTF8-character
    case 0xC2: return  (ascii);  break;
    case 0xC3: return  (ascii | 0xC0);  break;
    case 0x82: if (ascii == 0xAC) return (0x80);    // special case Euro-symbol
  }

  return  0; // otherwise: return zero, if character has to be ignored
}
`
????????????????

pilnikov, 2018-04-11:
I know how to "conquer" this problem.
But how to add "IT" to the source code

lorol, 2018-04-11:
...
switch (last) { // conversion depnding on first UTF8-character

// --- add following lines start ---------- 0xD0 and 0xD1 : Cyrillic, thanks bangka for this hint
    case 0xD0: if (ascii == 0x81) return  (0xA8);   //Ё
               else if (ascii >= 0x90 && ascii <= 0xBF) return  (ascii + 0x30);
               break;
    case 0xD1: if (ascii == 0x91) return  (0xB8);   //ё
               else if (ascii >= 0x80 && ascii <= 0x8F) return  (ascii + 0x70);
               break;
// --- add following lines end ----------
      
    case 0xC2: return (ascii); break;
....

pilnikov, 2018-04-12:
[image]
[image]
Thanks lorol!!!!

lorol, 2018-04-13:
Very good! Enjoy!

pilnikov, 2018-04-14:
[https://github.com/pilnikov/Color-Weather-Station-ESP32/blob/master/esp32-weather-station-color-rus/src/pic1.jpg](url)

closed 2018-04-14 by pilnikov

techbv, 2019-09-27:
[image]

techbv, 2019-09-27:
No touch, no carusel.
[esp8266-weather-station-color_no_touch.zip](https://github.com/ThingPulse/esp8266-weather-station-color/files/3664074/esp8266-weather-station-color_no_touch.zip)

G6EJD, 2019-09-27:
I need to copy the source code I think.

Most likely your screen has no touch panel installed, many don’t as they are essentially fake or sold deliberately without a touch panel to get the cost down, can you see the touch panel grid?

techbv, 2019-09-27:
This is not a complaint. Here I shared the cropped version :)