ThingPulse/esp8266-weather-station-color #62: compile for ESP32 Lolin32 lite

blotfi, 2018-04-01:
Hi
I am trying to compile this prg to ESP32 Lolin32 lite
I installed, all the lib, modified to
#include <ESPWiFi.h>
added
#include "SPIFFS.h" in TouchControllerWS.cpp

but it still complain about
ESP.function_of_sleep_and_so...

also, how can I get rid off DST, there is no DST in my country.

#define UTC_OFFSET + 1
struct dstRule StartRule = {"CEST", Last, Sun, Mar, 2, 3600}; // Central European Summer Time = UTC/GMT +2 hours
struct dstRule EndRule = {"CET", Last, Sun, Oct, 2, 0};       // Central European Time = UTC/GMT +1 hour

thanks

blotfi, 2018-04-01:
I see that it is because I didn't connected the touch part that the carousel toggles the screen very fast!

G6EJD, 2018-04-01:
DST change 3600 to 0 then time won’t is +1 hr in secs.

blotfi, 2018-04-01:
I will delete the use of the touch screen since the ESP32 has a touch interface ;-)

closed 2019-06-08 by stale[bot]