ThingPulse/esp8266-weather-station-color #63: name of forecastIcon to public

pilnikov, 2018-04-18:
Please add one more method in `WundegroundClient` by analogy with `getTodayIconText`

in .h  (section public: )
   ` String getIconText(int period);`

in .cpp
`String WundergroundClient::getIconText(int period) {`
  `return forecastIcon[period];`
`}`

TNX

marcelstoer, 2018-04-18:
There's actually a pending PR for this https://github.com/ThingPulse/esp8266-weather-station/pull/59 -> we'll track it there

closed 2018-04-18 by marcelstoer

G6EJD, 2018-04-18:
Some of those Icon descriptions can be (relatively) long, at least 12 and up to 16 chars and with screen space limited there seems little point in including this within the core programme and IMO it starts to degrade the advantage of the less is more display as is now.

pilnikov, 2018-04-20:
This is not for draw on display, it's for the icon selection

G6EJD, 2018-04-20:
There's already a variable called  'iconText' of type String and it's used for icon selection. e.g.
 if (iconText == "chanceflurries") return "X";

G6EJD, 2018-04-20:
From the ESP8266 weather station have you tried compiling it with that variable e.g. serial.print(iconText)

pilnikov, 2018-04-20:
dual conversion - first: from icon name to symbol; after: from symbol to "iconText" variable.

G6EJD, 2018-04-20:
You have to trace it through, the includes are relatively complex/convoluted but it's all there.