ThingPulse/esp8266-weather-station-color #64: true 4 bit color

pilnikov, 2018-04-29:
I use icons as a bmp file loaded into spiffs. The color of the icons and font is not displayed correctly. The settings indicate BitPerPixel = 4. Also i change the palette[]. But the screen displays only yellow blue white black and gray. 
I made a correction for the palette in minigrafx. [https://github.com/ThingPulse/minigrafx/issues/8] After that, the color of the moon changed from blue to gray, but the color of the sun changed from blue to yellow, although it should be red. Also, if I try to specify a text color such as Cyan, instead I get only either yellow or blue.

G6EJD, 2018-04-29:
You need at least 8-bit colour depth or there will always be a mismatch.

pilnikov, 2018-04-29:
if I just change the BitPerPixel to 8, then some of the picture is scattered, but the color of the sun on the icon remains yellow

pilnikov, 2018-04-29:
'tft.writePixel(tft.color565(r,g,b));'
This is the line from the example spitftbitmap.ino from adafruit gfx library (from which the drawbmpfromfile method from minigrafx was copied.) I can just set the color as well as there.

G6EJD, 2018-04-29:
OK, so 16-bit colour images required (5R,6G,5B) are your Icons 565 format?

pilnikov, 2018-04-30:
no, icon color dept is 24bit, its a rgb888 format. I try convert 565 to 888 and 888 to 565 - colors not correct

G6EJD, 2018-04-30:
I checked the code, so (drawBmpFromFile(String filename, uint8_t xMove, uint16_t yMove) ) requires a 24-bit color image coded bottom to top. 

What I would do is to write a small programme that does nothing other than display an image from your SD / SPIFFS FS? This will then enable you to narrow down where the problem is occurring.

If it draws incorrectly and assuming it works - it does, then your image format is incorrect.

What display are you using ILI9341?

pilnikov, 2018-04-30:
Yes 9341, I try it with adafruit gfx - it's ok! I download icon from squix blog. Color incorrectly not only in icon, but text color also incorrect. Problems in palette, I think

pilnikov, 2018-04-30:
i use the old version minigraf (with true color) on esp8266 all work fine. Now i use the ESP32  but my picture does not please me. Everything would be great if I dont know the old version.

G6EJD, 2018-04-30:
So it works on the ESP8266 and you've ported it across to the ESP32, but if the code is identical(?) apart from some mods to ticker and Wi-Fi connections, the results should be the same?

pilnikov, 2018-04-30:
everything works except for color

pilnikov, 2018-04-30:
I can use adafruit but there will be difficulties with fonts and some minigrafx-specific methods (e.g. progress bar)

G6EJD, 2018-04-30:
That’s what I mean it’s the same code so there has to be a difference with one of the includes and it must be pulling in the newer minigrafx? Have you turned on the verbose compiler option and checked what file it is including I think it’s a new one

pilnikov, 2018-04-30:
It was the old version that was normal in terms of color (it is very different from the new one, it was closer to adafruit gfx. "but the memory may be short, so Daniel used 4-bit changes and changing the palette"

closed 2019-06-08 by stale[bot]