ThingPulse/esp8266-weather-station-color #65: First and last forecast icons not showing properly, and the icons/description for the daily weather are incorrect.

sethpark777, 2018-05-03:
After installing and using the weather station for a little while, I have noticed a couple different things. 

1. When you look at the forecast on the scroller, the forecast for the current day always has an "unknown" icon. And when you look at the forecast details (not on the main screen), the current day forecast icon is unknown, and the last one is unknown. For example, today is Wednesday, and if i look at the forecast details, Wednesday will be unknown, and Monday night will be unknown. 

2. The Icons/ daily forcast summary seems to be wrong. When looking at the forecast, you will see a high and low temp, and the percentage of precipitation that correlates properly with the JSON that is returned from WU, but, the icons and forecast summary on the weather station are not consistent with the JSON from WU. 

If there is anything I can do to help clarify what I am seeing, please let me know and I will.

marcelstoer, 2018-05-03:
The first looks like a dup of #39 to me.

pilnikov, 2018-05-03:
[image]
[image]
[image]
[image]
[image]
[image]
Short name of dayofweek in json file is "Чт", on screen - "Ч" only. Second symbol "т" is lost. If chose "EN" lang - all ok.

pilnikov, 2018-05-03:
It seems to me that the mistake is that the number of periods x2 than the days of the week

sethpark777, 2018-05-03:
@marcelstoer That is what I originally thought, that it was the same issue, but after looking into it, the code that I have seems to be up to date with the changes mentioned in that thread (in regards to WundergroundForecast.cpp). There could be something I am missing though. 

@pilnikov I do not see anything like that in mine, I am using "EN" as my language. For your second comment, are you saying that I should try that? My current code does not look like that. When I do that, it does fix the first (current day) forecast Item and description. But, it does not fix the icon for the last day. 

Here is what I am talking about with the JSON not matching teh forecast Icons/descriptions. 
[image]
[image]
[image]
[image]

Maybe that will help sort things out?

pilnikov, 2018-05-03:
MAX_FORECASTS value make it is a 20

sethpark777, 2018-05-03:
@pilnikov that does seem to remedy the first issue, but the second remains. Why would changing the max forecast have an effect on the icons being displayed?

pilnikov, 2018-05-04:
In your original wundegroundforecast.cpp, the processing of the icons names  is incorrect. In json file 2 different views of the weather. 1 - txt forecast; 2 - simple forecast. In "txt" 20 periods (2 x 10 days as day/night). In "simple" 10 periods (only a day). For display the "night" period, you need "night" icons (their name starts with the prefix nt_), but default - in your forecasticon.h "night" icon is not present, and wundegroundforecast.cpp takes the name for icons from "simple", where twice less than required.
If MAX_FORECASTS = 12 your station displays only 6 periods correctly (6 x 2= 12, because "day/night") Therefore the last (seventh) is not correct.

marcelstoer, 2018-05-04:
Support for night-time icons should be isolated from this. How this can/should be implemented properly(?) is outlined at https://github.com/ThingPulse/espaper-weatherstation/issues/9.

Fixes for `wundegroundforecast.cpp` should be raised as PRs for the library repo at https://github.com/ThingPulse/esp8266-weather-station, thanks.

sethpark777, 2018-05-04:
So essentially the icons are being pulled from the simpleforecast vs the txt_forecast? Should I move this issue to https://github.com/ThingPulse/esp8266-weather-station to continue discussion? After applying the changes to remove the prefix for night time icons, it is still displaying the incorrect icons.

sethpark777, 2018-05-04:
Update: I added some logging in the `WundergroundForecast.cpp` to see what was going on. From my point of view (as someone that is not familiar with the codebase at all), everything is going correctly until it starts evaluating the json from the simpleforecast. When it hits the second if statement, that specifies the simpleforecast, it seems to overwrite what has already been set from the forecast_txt. 

Which makes me wonder, what is the point of the simpleforecast? unless it is being utilized to extract the highs and lows easier. 

That may have been what you were trying to say earlier essentially.

G6EJD, 2018-05-04:
Simpleforecast gives a forecast AM/PM for the next 3-days. Example:
		"simpleforecast": {
		"forecastday": [
		{"date":{
	"epoch":"1523728800",
	"pretty":"7:00 PM BST on April 14, 2018",
	"day":14,
	"month":4,
	"year":2018,
	"yday":103,
	"hour":19,
	"min":"00",
	"sec":0,
	"isdst":"1",
	"monthname":"April",
	"monthname_short":"Apr",
	"weekday_short":"Sat",
	"weekday":"Saturday",
	"ampm":"PM",
	"tz_short":"BST",
	"tz_long":"Europe/London"
},
		"period":1,
		"high": {
		"fahrenheit":"66",
		"celsius":"19"
		},
		"low": {
		"fahrenheit":"46",
		"celsius":"8"
		},
		"conditions":"Chance of Rain",
		"icon":"chancerain",
		"icon_url":"http://icons.wxug.com/i/c/k/chancerain.gif",
		"skyicon":"",
		"pop":40,
		"qpf_allday": {
		"in": 0.00,
		"mm": 0
		},
		"qpf_day": {
		"in": null,
		"mm": null
		},
		"qpf_night": {
		"in": 0.00,
		"mm": 0
		},
		"snow_allday": {
		"in": 0.0,
		"cm": 0.0
		},
		"snow_day": {
		"in": null,
		"cm": null
		},
		"snow_night": {
		"in": 0.0,
		"cm": 0.0
		},
		"maxwind": {
		"mph": 9,
		"kph": 14,
		"dir": "",
		"degrees": 0
		},
		"avewind": {
		"mph": 0,
		"kph": 1,
		"dir": "South",
		"degrees": 169
		},
		"avehumidity": 84,
		"maxhumidity": 0,
		"minhumidity": 0
		}
		,
		{"date":{
	"epoch":"1523815200",
	"pretty":"7:00 PM BST on April 15, 2018",
	"day":15,
	"month":4,
	"year":2018,
	"yday":104,
	"hour":19,
	"min":"00",
	"sec":0,
	"isdst":"1",
	"monthname":"April",
	"monthname_short":"Apr",
	"weekday_short":"Sun",
	"weekday":"Sunday",
	"ampm":"PM",
	"tz_short":"BST",
	"tz_long":"Europe/London"
},
		"period":2,
		"high": {
		"fahrenheit":"54",
		"celsius":"12"
		},
		"low": {
		"fahrenheit":"43",
		"celsius":"6"
		},
		"conditions":"Chance of Rain",
		"icon":"chancerain",
		"icon_url":"http://icons.wxug.com/i/c/k/chancerain.gif",
		"skyicon":"",
		"pop":80,
		"qpf_allday": {
		"in": 0.13,
		"mm": 3
		},
		"qpf_day": {
		"in": 0.13,
		"mm": 3
		},
		"qpf_night": {
		"in": 0.00,
		"mm": 0
		},
		"snow_allday": {
		"in": 0.0,
		"cm": 0.0
		},
		"snow_day": {
		"in": 0.0,
		"cm": 0.0
		},
		"snow_night": {
		"in": 0.0,
		"cm": 0.0
		},
		"maxwind": {
		"mph": 20,
		"kph": 32,
		"dir": "SSE",
		"degrees": 166
		},
		"avewind": {
		"mph": 15,
		"kph": 24,
		"dir": "SSE",
		"degrees": 166
		},
		"avehumidity": 91,
		"maxhumidity": 0,
		"minhumidity": 0
		}
		,
		{"date":{
	"epoch":"1523901600",
	"pretty":"7:00 PM BST on April 16, 2018",
	"day":16,
	"month":4,
	"year":2018,
	"yday":105,
	"hour":19,
	"min":"00",
	"sec":0,
	"isdst":"1",
	"monthname":"April",
	"monthname_short":"Apr",
	"weekday_short":"Mon",
	"weekday":"Monday",
	"ampm":"PM",
	"tz_short":"BST",
	"tz_long":"Europe/London"
},
		"period":3,
		"high": {
		"fahrenheit":"55",
		"celsius":"13"
		},
		"low": {
		"fahrenheit":"48",
		"celsius":"9"
		},
		"conditions":"Mostly Cloudy",
		"icon":"mostlycloudy",
		"icon_url":"http://icons.wxug.com/i/c/k/mostlycloudy.gif",
		"skyicon":"",
		"pop":10,
		"qpf_allday": {
		"in": 0.00,
		"mm": 0
		},
		"qpf_day": {
		"in": 0.00,
		"mm": 0
		},
		"qpf_night": {
		"in": 0.00,
		"mm": 0
		},
		"snow_allday": {
		"in": 0.0,
		"cm": 0.0
		},
		"snow_day": {
		"in": 0.0,
		"cm": 0.0
		},
		"snow_night": {
		"in": 0.0,
		"cm": 0.0
		},
		"maxwind": {
		"mph": 25,
		"kph": 40,
		"dir": "SSW",
		"degrees": 210
		},
		"avewind": {
		"mph": 17,
		"kph": 27,
		"dir": "SSW",
		"degrees": 210
		},
		"avehumidity": 71,
		"maxhumidity": 0,
		"minhumidity": 0
		}
		,
		{"date":{
	"epoch":"1523988000",
	"pretty":"7:00 PM BST on April 17, 2018",
	"day":17,
	"month":4,
	"year":2018,
	"yday":106,
	"hour":19,
	"min":"00",
	"sec":0,
	"isdst":"1",
	"monthname":"April",
	"monthname_short":"Apr",
	"weekday_short":"Tue",
	"weekday":"Tuesday",
	"ampm":"PM",
	"tz_short":"BST",
	"tz_long":"Europe/London"
},
		"period":4,
		"high": {
		"fahrenheit":"59",
		"celsius":"15"
		},
		"low": {
		"fahrenheit":"46",
		"celsius":"8"
		},
		"conditions":"Partly Cloudy",
		"icon":"partlycloudy",
		"icon_url":"http://icons.wxug.com/i/c/k/partlycloudy.gif",
		"skyicon":"",
		"pop":10,
		"qpf_allday": {
		"in": 0.00,
		"mm": 0
		},
		"qpf_day": {
		"in": 0.00,
		"mm": 0
		},
		"qpf_night": {
		"in": 0.00,
		"mm": 0
		},
		"snow_allday": {
		"in": 0.0,
		"cm": 0.0
		},
		"snow_day": {
		"in": 0.0,
		"cm": 0.0
		},
		"snow_night": {
		"in": 0.0,
		"cm": 0.0
		},
		"maxwind": {
		"mph": 30,
		"kph": 48,
		"dir": "SSW",
		"degrees": 200
		},
		"avewind": {
		"mph": 21,
		"kph": 34,
		"dir": "SSW",
		"degrees": 200
		},
		"avehumidity": 78,
		"maxhumidity": 0,
		"minhumidity": 0
		}
		]
		}
	}
		,	"moon_phase": {
		"percentIlluminated":"2",
		"ageOfMoon":"28",
		"phaseofMoon":"Waning Crescent",
		"hemisphere":"North",
		"current_time": {
		"hour":"20",
		"minute":"03"
		},
		"sunrise": {
		"hour":"6",
		"minute":"15"
		},
		"sunset": {
		"hour":"20",
		"minute":"03"
		},
		"moonrise": {
		"hour":"6",
		"minute":"04"
		},
		"moonset": {
		"hour":"18",
		"minute":"07"
		}
	},
	"sun_phase": {
		"sunrise": {
		"hour":"6",
		"minute":"15"
		},
		"sunset": {
		"hour":"20",
		"minute":"03"
		}
	}
}

sethpark777, 2018-05-04:
@G6EJD that is true of the `forecast` api, but the `WundergroundForecast.cpp` uses the `forecast10day` api.

pilnikov, 2018-05-05:
I corrected myself. Cut nt_ from the "night" icons. Changed in the request to WU from 10 days to 7. In the carousel change the first day from  today to tomorrow. max_forecast set 14. The detailed windows  also made from tomorrow (2-4-6 and 8-10-12 instead of 0-2-4 6-8-10).
See here ThingPulse/espaper-weatherstation#9.

G6EJD, 2018-05-05:
'@G6EJD that is true of the forecast api (it displays 4 days, current + next 3, like in your JSON), but the WundergroundForecast.cpp uses the forecast10day api.'

Agreed, but only 'forecast' has the 'simpleforecast' entries, there seems to be up to 4 calls being made:
current
forecast
forecast10day
astronomy

sethpark777, 2018-05-05:
@pilnikov I will try to mess around with the different numbering of days. I removed the if statement:
`if (currentKey == "icon" && isSimpleForecast && currentForecastPeriod < maxForecasts) {
	  Serial.printf("simpleforecast.");
    forecasts[currentForecastPeriod].forecastIcon = value;`
and that has seemed to make everything work correctly, even with my max_forecast set to 20. 

@G6EJD unless I am missing what you are saying, the `forecast` API call is not the only one that contains `simpleforecast` entries. Take a look at the `forecast10day` json that I have attached. 
[WU10DayForecast.txt](https://github.com/ThingPulse/esp8266-weather-station-color/files/1976730/WU10DayForecast.txt)

G6EJD, 2018-05-05:
Thanks I missed the inclusion of the simpleforecast in the 10day variant.

G6EJD, 2018-05-06:
I thought I'd mention some of WU API response anomalies I've noted:
icon names e.g. for 'clear' can also be returned as 'Clear' so now I use icon.toLowerCase(); to solve that and in fact all responses where a comparison is made can usefully have that added. 
Wind direction responses occasionally revert to full text for cardinal points:
N
NE
E
SE
S
SW
W
NW
Can also be:
North
NE
East
SE
South
SW
West
NW
So I cater for both returns ("N" || "North")

marcelstoer, 2018-05-06:
@G6EJD would you be so kind to create a PR for the weather station lib at https://github.com/ThingPulse/esp8266-weather-station?

G6EJD, 2018-05-06:
Do you mean for inconsistent WU API responses? I haven't created a branch of the code it's what I do in my versions of WU reading code. 
There is also the incorrect Moon display (New Moon is shown with 2% illumination) and adding North/South hemisphere detection, which I've done, but not had time to test it and it needs the moon fonts to be added to, because position A should have been a New Moon - all black circle then B becomes what is now A and so on, if that makes sense.

marcelstoer, 2018-05-06:
> Do you mean for inconsistent WU API responses?

Well, whatever fix you have in _your_ code that would improve _this_ code 😜From your comments I conclude that you already fixed/improved some of the issue mentioned here (pity this thread is so chaotic).

marcelstoer, 2019-02-19:
[Weather Underground closed their existing API](https://apicommunity.wunderground.com/weatherapi/topics/end-of-service-for-the-weather-underground-api) for non-paying customers on 2018-12-31. After [they stopped issuing free API keys](https://thingpulse.com/hello-openweathermap-bye-bye-wunderground/) for low-volume personal use in May we migrated to OpenWeatherMap and [deprecated all WU code](https://github.com/ThingPulse/esp8266-weather-station#deprecation-notes) in this library.

We are now closing all WU-related issues and PRs.

closed 2019-02-19 by marcelstoer