ThingPulse/esp8266-weather-station-color #66: WGHourly copy paste bug

hjcschaefer, 2018-05-09:
Use one of the two templates below and delete the rest.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
This code does not compile
```c
typedef struct WGHourly {
  String icon;
  String title;
  String temp;
  String hour;
  String PoP;
} WGForecast;
```

### Actual behavior

Need to change WGForecast to WGHourly, looks like copy paste bug.

### Test code

```c
// add code here
```
### Weather Station Color version
clean checkout
### Hardware
Describe whether you run the code on the ThingPulse Weather Station Color kit or on some custom hardware.

8<------------------------ END BUG REPORT -------------------------------------

marcelstoer, 2018-05-09:
Thanks but this code is not part of this repository it's part of the library at https://github.com/ThingPulse/esp8266-weather-station -> should be reported there. That being said I can't personally reproduce this.

closed 2018-05-09 by marcelstoer

marcelstoer, 2018-05-09:
> sorry for accidentally creating the issue in the wrong repo.

No worries, it's just that we can't handle the issue here properly when code in the other repo is affected. Just checked and realized we have a pending PR over there: https://github.com/ThingPulse/esp8266-weather-station/pull/103

> remember basic C to see that this is a bug.

I wasn't saying that this be not a bug - just that if you checkout _this_ repo it will actually compile.

> wind speed and direction

Hmm, this rings a bell...I closed https://github.com/ThingPulse/esp8266-weather-station/issues/14 the other day because it wasn't complete. Would you mind putting your improvements up as a PR?

> Will not bother you again.

You're not bothering us.