ThingPulse/esp8266-weather-station-color #67: Sunny Icon with 100% rain

hjcschaefer, 2018-05-09:
### Expected behavior
When 100% rain is expected you would like to see the rain icon

### Actual behavior
Full sun icon was shown, see picture below. I got the json from around that time and this showed the correct icon name: 'rain'

### Test code
I could not attach the json but here is the snippet:
```c
{
		"period":4,
		"icon":"rain",
		"icon_url":"http://icons.wxug.com/i/c/k/rain.gif",
		"title":"Thursday",
		"fcttext":"Rain likely. High 57F. Winds WSW at 10 to 15 mph. Chance of rain 100%. Rainfall near a half an inch.",
		"fcttext_metric":"Periods of rain. High 14C. Winds WSW at 15 to 25 km/h. Chance of rain 100%. Rainfall around 12mm.",
		"pop":"100"
		}
```

### Weather Station Color version
Which branch are you on? If you know the Git revision then add it here as well.

### Hardware
bought a few weeks ago
[image]

G6EJD, 2018-05-09:
My observations are that Weather Underground do get things wrong, and at that instance they could well have been issuing an icon name of 'clear' or 'sunny'.

There is also a possibility as I have seen that they issue 'Rain' and not 'rain' which leads to other issues. 

A test that sets the icon name to 'rain' would prove if this is an issue or not.

marcelstoer, 2019-01-22:
[Weather Underground closed their existing API](https://apicommunity.wunderground.com/weatherapi/topics/end-of-service-for-the-weather-underground-api) for non-paying customers on 2018-12-31. After [they stopped issuing free API keys](https://thingpulse.com/hello-openweathermap-bye-bye-wunderground/) for low-volume personal use in May we migrated to OpenWeatherMap and [deprecated all WU code](https://github.com/ThingPulse/esp8266-weather-station#deprecation-notes) in this library.

We are now closing all WU-related issues and PRs.

closed 2019-01-22 by marcelstoer