ThingPulse/esp8266-weather-station-color #68: failure to compile due to missing declaration for drawWifiQuality

liamkennedy, 2018-05-14:
Use one of the two templates below and delete the rest.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
It should compile

### Actual behavior
Compilation fails with this error message

```
exit status 1
'drawWifiQuality' was not declared in this scope
```

### Test code
No code needed - it just fails to compile

### Weather Station Color version
The current version here on github (I cloned the project)

### Hardware
Using it on your kit - as purchased via Amazon.

8<------------------------ END BUG REPORT -------------------------------------

The fix is to add 
```
void drawWifiQuality();
````
after line 113 void drawSeparator(uint16_t y);

G6EJD, 2018-05-14:
The code does compile ok and these errors are always due to an errant extra character in the body somewhere before the section you have highlighted. Have you check the compiler error messages to see where the error first appears? Maybe the code has been modified in some way?

marcelstoer, 2018-05-14:
I can't reproduce this either but it's true that `drawWifiQuality` is the only function that isn't declared in the 102-117 section.

liamkennedy, 2018-05-14:
Hmm... well.. I made no changes to the core function prior to the compile (just updated setting.h as needed)

Adding that declaration in has everything work.  I can also comment out the call to drawWiFiQuality and it all works just fine too.  

I am running with the Arduino IDE on Windows 10 with the 1.9.0 BETA version.  There are some oddities with the release version of the Windows Arduino IDE that required use of the BETA version - so perhaps this is some add-on error related to the BETA IDE.

liamkennedy, 2018-05-14:
Just for completeness here is the initial reporting of the error in the compiler output.  

```
C:\Users\liamk\OneDrive\Documents\GitHub\esp8266-weather-station-color\esp8266-weather-station-color.ino: In function 'void drawTime()':

esp8266-weather-station-color:343:19: error: 'drawWifiQuality' was not declared in this scope

   drawWifiQuality();
```
No further clue as to any other errors encountered... and.. as I say.. it's all working now with the added declaration I described.

closed 2018-05-15 by marcelstoer

liamkennedy, 2018-05-16:
Thanks!  I appreciate the work you (all) do with this project.

commit e5091578 (2018-05-15): Add declaration for drawWifiQuality before use

Some compilation environements report errors if function isn't delcared.

Fixes #68