ThingPulse/esp8266-weather-station-color #69: Make full translation easier

tigers75, 2018-05-22:
8<------------------------ FEATURE REQUEST ------------------------------------
### More info on how to translate the interface, maybe with some string moved to the settings.h file

### Justification
I'd like to translate the whole interface to Italian (or any language): setting the proper language code in the settings for WU does the trick for 90% of the interface, but some stings remain in english and others are buried deep in one of the libraries used and not easy to find.

### Workarounds
I can see that some of the strings are in the main code in esp8266-weather-station-color.ino, and they can be edited in place, but it'd be much better to have them collected in the settings.h file.

I'haven't found yet month/ weekday names that must be in some library. I have a custom version of the Time library from the Arduino Core where I changed month and weekdays long and short, but it clearly picks the names from somewhere else.

8<------------------------ END FEATURE REQUEST --------------------------------

G6EJD, 2018-05-22:
The month and day text comes from WU, if you select IT for your language, does it not give you month / day in Italian, I thought WU then output in your language, maybe not.

tigers75, 2018-05-22:
It's only partially true: on top of the clock there're still english short text (see image). Also in the bottom part "Sun" and "Moon" are in english (but they're in the main 
[image]
code and easy to replace).

pilnikov, 2018-05-22:
https://github.com/pilnikov/Color-Weather-Station/blob/master/resources/pic1.jpg
all string on russian languange

marcelstoer, 2018-05-22:
Dup of #34

closed 2018-05-22 by marcelstoer