ThingPulse/esp8266-weather-station-color #72: Name of days and forecast detail disappear.

tigers75, 2018-05-26:
Use one of the two templates below and delete the rest.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior

### Actual behavior

### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.

```c
// add code here
```
### Weather Station Color version
Which branch are you on? If you know the Git revision then add it here as well.

### Hardware
Describe whether you run the code on the ThingPulse Weather Station Color kit or on some custom hardware.

8<------------------------ END BUG REPORT -------------------------------------

8<------------------------ FEATURE REQUEST ------------------------------------
### Missing feature

### Justification
Tell us why you would like to see this feature added.

### Workarounds
Are there any workarounds you currently have in place because the feature is missing?

8<------------------------ END FEATURE REQUEST --------------------------------

tigers75, 2018-05-30:
OK, I digged deeper in the weekend and it seems in fact that the missing data is not definitely gone, sometimes it shows up again after an update.
Looks like the problem is in the weather underground part: if some data is "missing" in parsing the WU response the display shows empty places. I think it should be modified to retry in a short while and in the meantime keep showing the old values when something goes wrong.

G6EJD, 2018-05-30:
I’ve been doing that in my weather station implementations and leave all the display data as last received until a valid new set of new data is received, although I think it makes for a better solution it’s often difficult to know if there’s been an update or not, so I’m divided and missing data on screen is a useful indicator of a problem. Now I’ve moved over to Openweathermap servers I’ve noticed they are more consistent and reliable.

tigers75, 2018-05-30:
Yes, I just noticed the change and I'm going to try it. Can you post the code you modified and maybe send a PR? I think it should just be there. Maybe if after some retries something is still wrong put a "not updated" message on screen.

marcelstoer, 2019-01-22:
[Weather Underground closed their existing API](https://apicommunity.wunderground.com/weatherapi/topics/end-of-service-for-the-weather-underground-api) for non-paying customers on 2018-12-31. After [they stopped issuing free API keys](https://thingpulse.com/hello-openweathermap-bye-bye-wunderground/) for low-volume personal use in May we migrated to OpenWeatherMap and [deprecated all WU code](https://github.com/ThingPulse/esp8266-weather-station#deprecation-notes) in this library.

We are now closing all WU-related issues and PRs.

closed 2019-01-22 by marcelstoer