ThingPulse/esp8266-weather-station-color #73: Available Pin for NEOPIXEL LED?

liamkennedy, 2018-05-30:
8<------------------------ FEATURE REQUEST ------------------------------------
### Available Pin for NEOPIXEL LED

### Justification
I'm looking to add a NEOPIXEL RGB LED strip to this project (I'm not using it for weatherstation although I do have a valid API key... yay me).  It appears the ONLY digital pin free is D0 - but from what I can gather (through testing) that PIN cannot be used to drive a NEOPIXEL.  

Testing without the TFT board hooked up I can use D8 / D1 / D2 (didn't try all of them)... but D0 just doesn't work.  Wemos datasheet says "All of the IO pins have interrupt/pwm/I2C/one-wire support except D0." so that may be why.  
 
I notice D8 is driving something called LED (the circuit trace seems to take it over to the TFT board).  Not sure if I should attempt to mess with any used pins... sounds like I would be asking for trouble.  

So really - I appreciate my question is not a Feature Request - but nevertheless I would be very grateful if anyone does have any suggestions for how I might add a Neopixel in to the mix with the hardware in this kit.  

Thank you so much!

8<------------------------ END FEATURE REQUEST --------------------------------

marcelstoer, 2018-05-30:
This is more about hardware then the software in this repository. This topic has been discussed here: https://support.thingpulse.com/502/which-pins-can-use-projects-with-your-weather-station-setup

closed 2018-05-30 by marcelstoer

liamkennedy, 2018-05-30:
Understood.. and really appreciate the link to that topic.