ThingPulse/esp8266-weather-station-color #74: Compile Error (some files missing)

ht81, 2018-06-01:
Use one of the two templates below and delete the rest.

8<------------------------ BUG REPORT -----------------------------------------
Can´t complie becouse of missing files. 
Missing files are:
#include <OpenWeatherMapCurrent.h>
#include <OpenWeatherMapForecast.h>
#include <Astronomy.h>

```c
// add code here
```
### Weather Station Color version
Which branch are you on? If you know the Git revision then add it here as well.

### Hardware
Describe whether you run the code on the ThingPulse Weather Station Color kit or on some custom hardware.

8<------------------------ END BUG REPORT -------------------------------------

8<------------------------ FEATURE REQUEST ------------------------------------
### Missing feature

### Justification
Tell us why you would like to see this feature added.

### Workarounds
Are there any workarounds you currently have in place because the feature is missing?

8<------------------------ END FEATURE REQUEST --------------------------------
Is it possbile to geht these files? Or does it takes some time because of bugfix wounderground api?

Greetz from Germany 
Thomas

G6EJD, 2018-06-01:
You need to refresh/download again:
https://github.com/ThingPulse/esp8266-weather-station
Then the files will be available, or download the same repository and copy into your current location.

closed 2018-06-01 by marcelstoer

ht81, 2018-06-01:
Thanks, but this link is for the ssd1306 display verion and not for the "ILI9341 Color Display TFT" that i use.... Or i am wrong?

marcelstoer, 2018-06-01:
You need that "ESP8266 WeatherStation" library as a dependency: https://docs.thingpulse.com/guides/wifi-color-display-kit/#install-libraries