ThingPulse/esp8266-weather-station-color #76: No data from Open Weather shown

ht81, 2018-06-04:
Use one of the two templates below and delete the rest.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
No data from Open Weather is shown on display

### Actual behavior

### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.

```c
// add code here
```
### Weather Station Color version
Which branch are you on? If you know the Git revision then add it here as well.

### Hardware
Describe whether you run the code on the ThingPulse Weather Station Color kit or on some custom hardware.

8<------------------------ END BUG REPORT -------------------------------------

8<------------------------ FEATURE REQUEST ------------------------------------
### Missing feature

### Justification
Tell us why you would like to see this feature added.

### Workarounds
Are there any workarounds you currently have in place because the feature is missing?

8<------------------------ END FEATURE REQUEST --------------------------------

I have succesfully created a API for Open Weather and copie it via copy & paste into settings.h. Also i have edit the "location". My problem ist i can´t see anything on the screen from open weather. 
It shown only a cloud with questionmark. Time / Touchscreen is working correct.on slidung forcast at page 1, the time on the little forcasts ist 2:00.... and nothing more infos about rain, temp etc....

what do i wrong?

tedbmoss, 2018-07-09:
Same problem

tedbmoss, 2018-07-09:
Using your hardware

marcelstoer, 2018-07-09:
If you don't provide the serial console output there's no way for us to assess this support case.

tedbmoss, 2018-07-10:
error is city not found, I was trying Peoria,AZ I tried Phoenix,US and it worked.

closed 2018-07-21 by marcelstoer

tedbmoss, 2018-07-21:
I just used the city code.

marcelstoer, 2018-07-21:
Yep, this issue inspired #81 and ultimately lead to ThingPulse/esp8266-weather-station#122 (using OWM location id will become new default).