ThingPulse/esp8266-weather-station-color #77: OpenWeatherMap includes missing

al-ang, 2018-06-20:
The sketch has the following includes
  #include <OpenWeatherMapCurrent.h> 
  #include <OpenWeatherMapForecast.h> 

but the files don't seem to be in the GitHub repository.  Will you publish them soon, or am I making an obvious error and don't see something that's right in front of my eyes?

G6EJD, 2018-06-20:
You need to install •ESP8266 Weather Station using Sketch, Manage search for •ESP8266 Weather Station install it, the libraries are in that repository. Or go to the •ESP8266 Weather Station repository on Github and have a look in the SRC folder there.

al-ang, 2018-06-20:
ah ... got it.  thank you!

closed 2018-06-20 by al-ang

marcelstoer, 2018-06-20:
The guide for customers - and everybody else - is linked on the README: https://github.com/ThingPulse/esp8266-weather-station-color#step-by-step-tutorial