ThingPulse/esp8266-weather-station-color #83: How to change to Greenwich Mean Time ?

nikki-reprap, 2018-07-12:
How can i change this to greenwich mean time as the displayed time is +1 from GMT
```
#define UTC_OFFSET +1
struct dstRule StartRule = {"CEST", Last, Sun, Mar, 2, 3600}; // Central European Summer Time = UTC/GMT +2 hours
struct dstRule EndRule = {"CET", Last, Sun, Oct, 2, 0};       // Central European Time = UTC/GMT +1 hour
```

G6EJD, 2018-07-12:
#define UTC_OFFSET +0 
struct dstRule StartRule = {"CEST", Last, Sun, Mar, 2, 0}; // Central European Summer Time = UTC/GMT +2 hours 
struct dstRule EndRule = {"CET", Last, Sun, Oct, 2, 0};       // Central European Time = UTC/GMT +1 hour

nikki-reprap, 2018-07-12:
Thanks using your example made the time -1 GMT but it helped me to understand what to change.

 ```
 #define UTC_OFFSET +1
struct dstRule StartRule = {"CEST", Last, Sun, Mar, 2, 0}; // Central European Summer Time = UTC/GMT +2 hours 
struct dstRule EndRule = {"CET", Last, Sun, Oct, 2, 0};       // Central European Time = UTC/GMT +1 hour 
```

closed 2018-07-21 by marcelstoer