ThingPulse/esp8266-weather-station-color #84: Using without ILI9341, but without touch - permanently switching between 12/24 hour

Schawen, 2018-07-23:
Hey,
I connected Wemos D1 min Pro to ILI9341 LCD without touch IC.
Problem:  
The display permanently switches ( two times per sec ) between 12 hour and 24 hour time display.
The forecast also gets updated in the same interval.

Is that behaviour a result of the missing touch ic on the back of the display board?
How can I avoid this behaviour?

Hope somebody can help ...

G6EJD, 2018-07-23:
You've got a non-standard display and so the code is not supported for your configuration. You would need to disable all function calls associated with the touch-panel and would require a considerable amount of work. You could try searching other forums to see if someone has produced a version without the touch panel.

Or you could check out line 39 in Settings.h 
37 #define HAVE_TOUCHPAD 

Maybe change that to DONOT_HAVE_TOUCHPAD :0

Schawen, 2018-07-23:
Thanks for your reply.
I previously searched the source code for the defined "HAVE_TOUCHPAD", but it seems that it is not used in the code at all.
The Display works fine, even without touch function - except for the permanently switching between 12/24 hour time display.

amjm22, 2018-08-02:
Attached is the adjusted code that is working.  The 12-24 issue is gone. 

[final_weather.txt](https://github.com/ThingPulse/esp8266-weather-station-color/files/2253092/final_weather.txt)

marcelstoer, 2018-08-02:
I had a brief look at the diff (would be easier if this were in a PR) and two things look...let's say _interesing_:

line ~186 (yes, it does compile)
```
while (!touchController.isCalibrationFinished())
touchController.saveCalibration();
{
  gfx.fillBuffer(0);
```

line ~220, this is effectively a no-op
```
if (p.y < 80) {
  IS_STYLE_12HR = IS_STYLE_12HR; // was IS_STYLE_12HR = !IS_STYLE_12HR;
```

plazmax, 2018-12-10:
> 
> 
> Attached is the adjusted code that is working. The 12-24 issue is gone.
> 
> [final_weather.txt](https://github.com/ThingPulse/esp8266-weather-station-color/files/2253092/final_weather.txt)

thank you i'll try :)

Perimec, 2018-12-22:
Hi,
first thanks for the code... however I'm having the same issue even if I have touch screen display. Not sure why. by the way, what can you do with the touch? I have tried and it doesn't do anything... maybe there is the problem.

thanks in advance

TinemChou, 2019-01-09:
With a chinese ILI9341 display the touch pad is not working too, i use 

`#define HAVE_TOUCHPAD` in settings.h

to prevent the usage of the touch pad the on the source code, for example : 

`#ifdef HAVE_TOUCHPAD`
`#include "TouchControllerWS.h"`
`#endif`

I also simply replace the touch pad functionnality with a push on D3 (with the internal pull-up resistor) and always use 24H hour style.

Dancopy, 2019-06-04:
> Hey,
> I connected Wemos D1 min Pro to ILI9341 LCD without touch IC.
> Problem:
> The display permanently switches ( two times per sec ) between 12 hour and 24 hour time display.
> The forecast also gets [...]

Hi Schawen!
I have the same problem - did you solve it? If so, can you post the solution? Thanks

Dancopy, 2019-06-04:
> 
> 
> > Attached is the adjusted code that is working. The 12-24 issue is gone.
> > [final_weather.txt](https://github.com/ThingPulse/esp8266-weather-station-color/files/2253092/final_weather.txt)
> 
> thank you i'll try :)

You're showing this error:
esp8266-weather-station-color:290:81: error: 'OPEN_WEATHER_MAP_LOCATION' was not declared in this scope

exit status 1
'OPEN_WEATHER_MAP_LOCATION' was not declared in this scope

Dancopy, 2019-06-04:
> 
> 
> With a chinese ILI9341 display the touch pad is not working too, i use
> 
> `#define HAVE_TOUCHPAD` in settings.h
> 
> to prevent the usage of the touch pad on the source code, for example :
> 
> `#ifdef HAVE_TOUCHPAD`
> `#include "TouchControllerWS.h"`
> `#endif`
> 
> I also simply replace the touch pad functionnality with a push on D3 (with the internal pull-up resistor) and always use 24H hour style.

Hi TinemChou, could you kindly explain a little more detail how you did or, post your sketch here and also how you connected the Push Button on the D3 pin? I could not understand correctly; thank you

closed 2019-12-15 by stale[bot]