ThingPulse/esp8266-weather-station-color #85: Wrong time in display

jwende, 2018-08-04:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
Show correct time based on location setting e.g. 09:35

### Actual behavior
Shows a time 7hours in future ...e.g. 18:35

```
### Weather Station Color Version 
Latest master branch

### Hardware
On  ThingPulse Weather Station Color kit 
8<------------------------ END BUG REPORT -------------------------------------
![04aea16c-9c04-49b7-9358-a08c76b9a62a](https://user-images.githubusercontent.com/3383864/43677736-46018a22-9807-11e8-815c-ff8aad923b98.jpeg)

G6EJD, 2018-08-04:
Which country are you in and have you set the correct time zone in the programme?

marcelstoer, 2018-08-04:
It started as a support request at https://support.thingpulse.com/731/wrong-time-displayed -> track it there

closed 2018-08-04 by marcelstoer