ThingPulse/esp8266-weather-station-color #87: Sunrise & Sunset shown 02:00 all the tim

galluslongus, 2018-08-18:
Sorry, i don´t know if this is the right place for this question.

I´m on the last version of weather station color.  everything works fine except the sunrise and sunset.
they shown 02:00 all the time. on next tab, overview, it shows 0 for wind speed, 0 for wind dir and also 0 for visibility.

Anyone knows how to fix it?

Thank you..

G6EJD, 2018-08-18:
Has it ever worked?
Location is important as not all locations get a sunrise/sunset report. Try changing your location to e.g. London, UK and see if that works. I’m aware that many parts of the Southern Hemisphere apart from Australia are not supported for the extra data.

marcelstoer, 2018-08-18:
Sounds very familiar: https://github.com/ThingPulse/espaper-weatherstation/issues/15 (which points to https://github.com/ThingPulse/esp8266-weather-station/issues/131).

Did you check what OWM actually returns? Try this in the browser after you replaced the tokens http://api.openweathermap.org/data/2.5/weather?id=location-id&appid=API-key&units=metric&lang=en

galluslongus, 2018-08-18:
Should work for my town (near Hamburg, Germany):

{"coord":{"lon":10.23,"lat":53.48},"weather":[{"id":803,"main":"Clouds","description":"broken clouds","icon":"04d"}],"base":"stations","main":{"temp":19.76,"pressure":1021,"humidity":60,"temp_min":19,"temp_max":20},"visibility":10000,"wind":{"speed":5.7,"deg":220},"clouds":{"all":75},"dt":1534580400,"sys":{"type":1,"id":4883,"message":0.0026,"country":"DE","sunrise":1534565050,"sunset":1534617621},"id":7290243,"name":"Bergedorf","cod":200}

But with LONDON ID it´s the same...shown 02:00

G6EJD, 2018-08-18:
It’s a bug

marcelstoer, 2018-08-18:
This is fixed with ESP8266 Weather Station lib 1.6.2: https://github.com/ThingPulse/esp8266-weather-station/releases/tag/v1.6.2

It should become available in the Arduino library manager in the next few hours.

closed 2018-08-18 by marcelstoer

galluslongus, 2018-08-18:
Awesome, thanx