ThingPulse/esp8266-weather-station-color #89: no temperature (0 C) with 1.6.2 library

sorriso93, 2018-09-06:
Use one of the two templates below and delete the rest.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
Hello, still having no temperature (0 C) with library version 1.6.2 and related example of weather station color...

### Test code
Last example version (this august)

8<------------------------ END BUG REPORT -------------------------------------

sorriso93, 2018-09-07:
I suspect the problem is correlated to a slow internet connection... sometimes I get the temperature and when the connection is overloaded I get 0 C.... is it possible?

AnsgarSchmidt, 2018-09-14:
I have the same problem here