ThingPulse/esp8266-weather-station-color #91: OpenWeatherMapCurrent isn't parsed properly - Still no data

odeBuXTeR, 2018-10-01:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior

Hi all

Data from OpenWeatherMap API current weather have to be displayed on the top of screen with the actual temperature, and correct sunrise/sunset on the bottom.

### Actual behavior
When launching the Weather Station, I have a questionmark on the cloud on the top, temperature at 0.0°C and sunrise/sunset at 2:00

### Test code
I check the url of the API and all is OK when requesting http://api.openweathermap.org/data/2.5/weather?id=3036083&appid=myAPIkey&units=metric&lang=fr in webbrowser, it returns :

{"coord":{"lon":1.81,"lat":48.44},"weather":[{"id":800,"main":"Clear","description":"ciel dégagé","icon":"01n"}],"base":"stations","main":{"temp":9.5,"pressure":1027,"humidity":66,"temp_min":9,"temp_max":11},"visibility":10000,"wind":{"speed":2.1,"deg":330},"clouds":{"all":0},"dt":1538425800,"sys":{"type":1,"id":5614,"message":0.0045,"country":"FR","sunrise":1538373185,"sunset":1538415024},"id":3036083,"name":"Aunay-sous-Auneau","cod":200}

So, I try to launch the OpenWeatherMapCurrentDemo.ino to see what it returns in serial console :

Next Loop-Step: 2059:
Getting url: http://api.openweathermap.org/data/2.5/weather?id=3036083&appid=592a6d77db32c8cbe343d33755ff103d&units=metric&lang=fr
[HTTP] GET...
[HTTP] GET... code: 200
------------------------------------
lon: 0.000000
lat: 0.000000
weatherId: 0
main: 
description: 
icon: 
iconMeteoCon: 
temp: -159465800836912022475758179315756302336.000000
pressure: 61438
humidity: 239
tempMin: -159465800836912022475758179315756302336.000000
tempMax: -159465800836912022475758179315756302336.000000
windSpeed: -159465800836912022475758179315756302336.000000
windDeg: -159465800836912022475758179315756302336.000000
clouds: 254
observationTime: -17829890, full date: Sun Jun  8 15:15:10 1969
country: 
sunrise: -17829890, full date: Sun Jun  8 15:15:10 1969
sunset: -17829890, full date: Sun Jun  8 15:15:10 1969
cityName: 

---------------------------------------------------/

I do not understand clearly what appened with this "currentweather" data but I tied to display this on my weather station ... :(

Tested on 3 Wemos D1 mini and pro with the same issue.

No problem with the forecast, all data clearly interpreted ! Only the currentweather issue.

### Weather Station Color version
With all the latest libraries, try with 1.6.1 of weather station and same issue

### Hardware
Custom hardware, try with 3 different Wemos D1 mini and PRO (not clone) and the screen is the 2.4 TFT

8<------------------------ END BUG REPORT -------------------------------------

sorriso93, 2018-10-01:
Same hardware configuration and same problem sometimes. For sure when my connection is slow I face this problem...

G6EJD, 2018-10-01:
Try a different location like Paris and see if the problem persists

marcelstoer, 2018-10-02:
This is likely a duplicate of #89.

> try with 1.6.1 of weather station and same issue

1.6.2 was meant to fix some of theses issues.

odeBuXTeR, 2018-10-02:
HHi, thank for your replies. I'm searching to solve this problem for 2 months now. I know the issue  #89 and for me it's not a duplicate.
I have two internet connection here and the station NEVER display the current data. 1.6.1 and 1.6.2 have been installed with no success, the current weather demo always return wrong informations. Trying different locations not solving the problem too, I've try paris, dublin and others. Just uploading a fresh weather station with just wifi credentials and OWM ID giving same result. For me the data is not parced properly and I don't know how to check this. One more time, station NEVER display the current weather conditions

odeBuXTeR, 2018-10-23:
Hello all, nice job ! All working fine, many thanks !!!

odygr, 2019-01-15:
hello, ,
i also having the same issue. i used versions. 1.6.3,  1.6.4 and 1.6.5 but with no luck