ThingPulse/esp8266-weather-station-color #95: Cant compile

totalnooob, 2018-12-23:
### Expected behavior
To compile it
### Actual behavior
Doesnt compile 

### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.

Platformio

```c> Executing task in folder wemos d1 mini weather station: C:\Users\Riso\.platformio\penv\Scripts\platformio.exe run <

Processing d1_mini (platform: espressif8266; board: d1_mini; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 > WeMos D1 R2 & mini
HARDWARE: ESP8266 80MHz 80KB RAM (4MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 47 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <WeatherStation> 1.6.0 #758f05e
|   |-- <JsonStreamingParser> 1.0.5
|   |-- <Time> 1.5
|   |-- <ESP8266HTTPClient> 1.1
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <Bridge> 1.7.0
|-- <ESP8266WiFi> 1.0
|-- <JsonStreamingParser> 1.0.5
|-- <SPI> 1.0
|-- <simpleDSTadjust> 1.2.0
|   |-- <Time> 1.5
|-- <Mini Grafx> 1.0.0
|   |-- <SPI> 1.0
|-- <XPT2046_Touchscreen>
|   |-- <SPI> 1.0
Compiling .pioenvs\d1_mini\src\main.cpp.o
In file included from src\settings.h:21:0,
from src\main.cpp:25:
C:\Users\Riso\.platformio\lib\simpleDSTadjust_ID1276/simpleDSTadjust.h:16:0: warning: "SECS_PER_MIN" redefined [enabled by default]
#define SECS_PER_MIN  (60)
^
In file included from C:\Users\Riso\.platformio\lib\Time_ID44/time.h:1:0,
from C:\Users\Riso\.platformio\lib\simpleDSTadjust_ID1276/simpleDSTadjust.h:12,
from src\settings.h:21,
from src\main.cpp:25:
C:\Users\Riso\.platformio\lib\Time_ID44/TimeLib.h:67:0: note: this is the location of the previous definition
#define SECS_PER_MIN  (60UL)
^
In file included from src\settings.h:21:0,
from src\main.cpp:25:
C:\Users\Riso\.platformio\lib\simpleDSTadjust_ID1276/simpleDSTadjust.h:17:0: warning: "SECS_PER_HOUR" redefined [enabled by default]
#define SECS_PER_HOUR (3600)
^
In file included from C:\Users\Riso\.platformio\lib\Time_ID44/time.h:1:0,
from C:\Users\Riso\.platformio\lib\simpleDSTadjust_ID1276/simpleDSTadjust.h:12,
from src\settings.h:21,
from src\main.cpp:25:
C:\Users\Riso\.platformio\lib\Time_ID44/TimeLib.h:68:0: note: this is the location of the previous definition
#define SECS_PER_HOUR (3600UL)
^
In file included from src\settings.h:21:0,
from src\main.cpp:25:
C:\Users\Riso\.platformio\lib\simpleDSTadjust_ID1276/simpleDSTadjust.h:18:0: warning: "SECS_PER_DAY" redefined [enabled by default]
#define SECS_PER_DAY  (SECS_PER_HOUR * 24)
^
In file included from C:\Users\Riso\.platformio\lib\Time_ID44/time.h:1:0,
[... 766 lines cut ...]
```c
Build options changed, rebuilding all
sketch\esp8266-weather-station-color.ino.cpp.o: In function `drawAbout()':

C:\Users\Riso\Desktop\esp8266-weather-station-color/esp8266-weather-station-color.ino:610: undefined reference to `gmtime'

sketch\esp8266-weather-station-color.ino.cpp.o: In function `getTime(long*)':

C:\Users\Riso\Desktop\esp8266-weather-station-color/esp8266-weather-station-color.ino:610: undefined reference to `gmtime'

libraries\ESP8266_Weather_Station\OpenWeatherMapForecast.cpp.o:(.text._ZN22OpenWeatherMapForecast5valueE6String+0x9e): undefined reference to `gmtime'

collect2.exe: error: ld returned 1 exit status

Multiple libraries were found for "ESP8266HTTPClient.h"
 Used: C:\Users\Riso\Documents\Arduino\libraries\ESP8266HTTPClient
 Not used: C:\Users\Riso\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266HTTPClient
exit status 1
Error compiling for board WeMos D1 R2 & mini.
```
### Weather Station Color version
Which branch are you on? If you know the Git revision then add it here as well.

### Hardware
Wemos D1 Mini, 2,8" TFT SPI 240x320 displa, Using latest code from github

G6EJD, 2018-12-24:
Did the authors say they supported PlatformIO? I think not.

totalnooob, 2018-12-24:
Oh didnt know it doesnt work on platformio and totally forget to update the esp library thanks

marcelstoer, 2018-12-24:
You can certainly use it with PlatformIO but you need to reorganize the code (at least put it into a `src` folder).

closed 2018-12-24 by marcelstoer