ThingPulse/esp8266-weather-station-color #96: How to change forecasts every hour interval?

Burn-In, 2019-01-24:
Currently the forcasts are every 12h. OWM 5 day forecast includes weather data every 3 hours.
How can i change the interval to 6 or 3 hours? I couldn't find anything nether in the scatch nor library.

Regards Joerg

Burn-In, 2019-01-28:
OK, i got it.

The forecasts are filtered by the allowedHours[] array. In WeatherStationColor.ino it looks like that:
allowedHours[] = {12, 0}; //forecasts for 12:00 and 0:00 (UTC) only

For forecasts every 3 hours it should looks like that:
allowedHours[] = {0, 3, 6, 9, 12, 15, 18, 21}; //forecasts every 3h (UTC)

For me it looks like that now:
allowedHours[] = {6, 15}; //forecasts for 6:00 and 12:00 (UTC) only

Issue can be closed. Maybe it is an feature for settings.h.

Regards Joerg

closed 2019-01-28 by Burn-In

Burn-In, 2019-01-29:
Ok. Never done this before. But i will try it. :)

marcelstoer, 2019-01-29:
It may look complicated or daunting at first but it really isn't. The GitHub documentation is great but if you follow our summary at https://github.com/ThingPulse/esp8266-weather-station-color/blob/master/CONTRIBUTING.md you probably won't even need that.

not4unco, 2019-12-20:
@Burn-In  Would you share your code please? I like the idea of updates on the hour. bubbasguppies@gmail.com

thanks