ThingPulse/esp8266-weather-station-color #97: Carousel blinking

Burn-In, 2019-01-24:
The forecast carousel is blinking more or less. Please look at the video here: https://youtu.be/umS0VCTuwes

This issue disappears if i comment off some content on the screen e.g. currentWeather.icon, currentWeather.description or moonAgeImage.

Is there a memory problem?

regards Joerg

wilhelmzeuschner, 2019-02-03:
I recently experienced the same issue and was convinced that it was somehow related to the fact that I used a screen without touch. Apparently this isn't the case.
I managed to get rid of the blinking by setting the Frequency of the ESP8266 to 160Mhz. This worked both with and without the touchscreen!
Downgrading libraries din't work but I have been able to use my hardware (no touch) without problems previously (March 2018).

Burn-In, 2019-02-06:
Yes, it works for me also.
With 160MHz the speed for a transition increases. So i put these lines in setup()::

//Set the direction if the automatic transitioning
carousel.setAutoTransitionBackwards();                         //was necessary too, otherwise wrong direction

//Set the approx. time a transition will take
carousel.setTimePerTransition(1000);

marcelstoer, 2019-03-18:
This is confirmed to be caused by ESP8266 Arduino core 2.5. As per [our guide](https://docs.thingpulse.com/guides/wifi-color-display-kit/#install-the-esp8266-toolchain) this is not supported/recommended yet. All still works fine on 2.4.2.

@squix78 not sure if the fix for 2.5 would have to be in Mini Grafx or the sketch.

marcelstoer, 2019-03-21:
Mini Grafx v1.2.3 released earlier today contains a fix for this: https://github.com/ThingPulse/minigrafx/releases/tag/1.2.3

closed 2019-03-21 by marcelstoer