ThingPulse/esp8266-weather-station-color #100: Support for WIFIEsp.h

not4unco, 2019-03-01:
Use one of the two templates below and delete the rest.

8<------------------------ BUG REPORT ------------------
### Hardware
Describe whether you run the code on the ThingPulse Weather Station Color kit or on some custom hardware.

I recently purchased a "WEMOS" Mega from eBay. I'd like to be able to use the forecasting code with my board, but always get errors due to wifi.h. I believe the connection between the Mega and Esp8266 uses AT commands. While the 2.8" SPI screen is ok, I'd like to be able to use a 7" Parallel UTFT screen for weather information as well as being able to send my current weather data from various sensors to Open Weather Maps.

8<------------------------ END BUG REPORT -------------------------------------

G6EJD, 2019-03-01:
I would not call this an error/bug report! It’s more of a desire to use the code on a board it was never designed to.
Change WiFi.h to ESP8266Wifi.h it should turn red if spelt correctly.

G6EJD, 2019-03-02:
BTW when you program an ESP8266 you will overwrite the often supplied operating system that gives you AT commands.

not4unco, 2019-03-07:
Grrrr.....the Mega communicates with the esp on serial3 hdwr with AT commands to OWM.

closed 2019-09-17 by stale[bot]